jneilliii/OctoPrint-TPLinkSmartplug #267: [Feature Request]: Support Upload Monitoring without Print

jbienz, 2021-04-25:
**Is your feature request related to a problem? Please describe.**
This is similar to #248.

**Describe the solution you'd like**
I regularly upload a file then do several tweaks in OctoPrint before actually hitting print. This can be changing spools with SpoolManager or setting various things in OctoPrint. I would still very much like to have my printer warm up and my lights turn on the moment I upload, even though I'm not ready to print just yet. 

**Describe alternatives you've considered**
The alternative right now is to manually turn on the printer and lights.

**Additional context**
No additional context. I understand the reason it was designed the way it currently is, but I'd really appreciate an option that says whether "print immediately" is required. Thank you!

jneilliii, 2021-04-25:
So basically power on with upload but only start print if the option is enabled?

jbienz, 2021-04-26:
Yes, power on with upload regardless of whether the print box is checked in Prusa or Cura. I almost never check that box because it bypasses other plugins. For example, if that box is checked in Prusa or Cura then SpoolManager doesn't get a chance to ask me if I have the right spool selected. So I will personally probably never have that box checked, but others may still prefer to use it.

WilfulTravesty, 2021-09-06:
I would find this useful as well, so that my light comes on and I can verify the bed is clear before I start printing.

JBlau101, 2022-04-13:
I'd also love to have it turn on immediately upon uploading a file, regardless of the decision to print. Thanks for this awesome plugin!

jneilliii, 2022-04-14:
added with above commit to rc branch will be available for testing in the next release candidate version. closing due to duplicate of #190

PR #304 (2022-04-17): 1.0.2
* resolves issue related to shutdown monitoring not powering off devices due to SIGTERM, resolves #274
* fix timelapse race condition, resolves for #281
* add JSON response to on/off API commands
* add `Always on Upload` option to not require start print from slicer upload, resolves #190, resolves #267
* replace deprecated getBoolen with get_boolean
* remove duplicated bundling of fontawesome 5 if running on OctoPrint 1.5.0+
closed 2022-04-14 by jneilliii