jneilliii/OctoPrint-TPLinkSmartplug #268: [Feature Request]: Option to only power off if a printer is connected to Octoprint

jalabb, 2021-04-26:
**Is your feature request related to a problem? Please describe.**
I was printing from the SD card, not using octoprint. The plugin thought the printer was idle and turned off the plug, ruining my print (the Prusa Mini doesn't resume after a power loss).

**Describe the solution you'd like**
A checkbox with something like "Only power off if a printer is connected" -> see attached mockup

[image]

jneilliii, 2021-04-26:
I'm sorry this happened to you, I don't print from SD card and you're actually the first person to bring this up. I never thought about that scenario, but was curious if you started the print from the printer's controller directly or from OctoPrint? I'm fairly certain that OctoPrint knows when printing from SD.

jalabb, 2021-04-26:
No problem, it was only a Benchy ^^

I was doing some sanity checking and wanted to cut Octoprint out of the equation completely so I disconnected the printer from Octoprint and I started the print from the printer. Since the plugin had no input from the printer, after the idle timeout ran out it did its job and cut the power :/

jalabb, 2021-04-26:
Thanks for looking into it :)

> toggle switch in the sidebar

Silly me, I'll think about it next time.