jneilliii/OctoPrint-TPLinkSmartplug #270: [BUG]:

phodara, 2021-05-10:
Everything was working perfectly.  I set the plugin timeout to switch the power off 90 secs after print job was finished and it originally worked beautifully.  After the the most recent Octopi updates the plugin switches the printer off immediately after the job is finished and then shows the abort countdown timer, but by then there is nothing to abort.  I double checked the plugin settings and everything looks correct.

[image]

phodara, 2021-05-10:
> Can you please share the settings you have for the individual plug and
> please upload the plugin_tplinksmartplug_debug.log and octoprint.log file
> from the logging section of OctoPrint. You can just drag them into a
> comment here on github.

Thank you for your response.  I am in the middle of a 24hr print job, as
soon as its complete I will send you the info.

Thanks again,

Paul

phodara, 2021-05-13:
Thank you for following up on this issue here is the requested log files and a print screen of the settings:

[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6472918/octoprint.log)
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6472919/plugin_tplinksmartplug_debug.log)
[image]

jneilliii, 2021-05-13:
Idle target temp is way to high...

phodara, 2021-05-13:
Shouldn't the high idle temp prevent the printer from shutting off, I did this as a test?  Printer shuts off immediately after job.

jneilliii, 2021-05-13:
Actually,  I think that might trigger setting the temp to that which could cause firmware errors, which would trigger the error off, which bypasses waiting for temp.

phodara, 2021-05-14:
Ahh ok thanks, I lowered the temp

phodara, 2021-05-16:
Same issue. I lowered Idle Target Temperature to 80 deg and it shuts off printer immediately after print job completes even though Idle timeout is now 30 minutes.

jneilliii, 2021-05-16:
Can you please share a screenshot of the setting for the plug? Are you using gcode commands and telling the plug to turn off at the end of your gcode or in OctoPrint's GCODE Scripts section? Those will run instantly as well and bypass the idle timer. Typically you wouldn't necessarily want to use both of them together.

phodara, 2021-05-17:
Success! I think that was the issue.  Editing the profile settings I saw GCode Trigger was checked. I unchecked GCode Trigger and lowered the Idle Target Temperature and everything seems to be working now.  I am not sure what happened probably me making a change inadvertently or maybe something corrupted during an Octopi update.  It appears all is well again.
Thank you!

closed 2021-06-09 by github-actions[bot]