jneilliii/OctoPrint-TPLinkSmartplug #272: [BUG]:  Shutdown does not operate my HS100 olug

meltonpieman, 2021-06-18:
**Describe the bug**
I have set up the close on idle (after 1min and 50c for a trial) - the abort button/message comes up and disappears after the set time. The printer disconnects from Octoprint. However, the plug does not turn off. The lightning icon starts to rotate but does not stop. I can shut the plug down using the KASA app on my phone.
Also, the lightning icon does not operate the plug.

**Expected behavior**
I expected the power to cut off.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 Home fully up to date.
 - Browser: Firefox fully up to date.
 - OctoPrint Version 1.6.1 Python 3.7.3 Octopi 0.18.0
 - Plugin Version 1.0.1
 - TP-link firmware up to date

**Additional context**
My wifi hub is a Virgin Superhub 3. I have fixed the IPs for the plug and my Octopi. The hub gives the HS100 and address on the 2.4 band 192.168.0.22/24 I have tried this and 192.168.0.22 - neither work.

jneilliii, 2021-06-18:
The plug's ip would be `192.168.0.22` in the plugin's settings. The only restriction is that the pi has to be able to communicate with the plug. There have been previous reports of this getting blocked by a specific firmware version in the UK, which I have no control over. To debug this further, enable debug logging in the plugin's settings, restart OctoPrint and let it try to idle out again. If it still does not work, share your plugin_tplinksmartplug_debug.log and octoprint.log and screenshots of your settings please.

meltonpieman, 2021-07-03:
> The plug's ip would be `192.168.0.22` in the plugin's settings. The only restriction is that the pi has to be able to communicate with the plug. There have been previous reports of this getting [...]
------------
I thought that I had forwarded the files, but  obviously nothing got here. So ... here is another try
David
[octoprint-logs.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6758289/octoprint-logs.zip)
[image]

jneilliii, 2021-07-03:
Your log indicates that the plug is not reacting to the commands being sent to it. This could be a network issue or it could be a firmware issue with TPLink blocking local control.

```
[2021-07-03 12:19:12,622] DEBUG: Could not connect to 192.168.0.22.
```

closed 2021-07-26 by github-actions[bot]