jneilliii/OctoPrint-TPLinkSmartplug #273: [BUG]: Current (Amps) incorrect for 240V

shanejimmink, 2021-06-22:
**Describe the bug**
When viewed in the TP-Link Smart Plug control window, the graph shows the Current and Power usage (with history). The Amperage is incorrect as it reports approximately 1 A, with the total power approximately 128W. As I am on 240V this should be approximately 0.5A. 

Refering to the KASA App It shows the current Power Usage as approximately 128W.

**To Reproduce**
1. Setup TPLink Smart Plug
2. Connect to ip address and enable.
3. Go to TP-Link Smart Plug control panel
4. View Plot and mouse over the Current Usage plot to view value.

**Expected behavior**
The Current (Amps) used should be calculated correctly for 240V power supplies. IE. 0.5A for 120W on 240V

jneilliii, 2021-06-22:
I'm not actually sure on this one, because I pull the data direct from the device. The only math involved on the data pulled is for converting devices that report mA.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/e84264efb36b3c76bf235783cc735a874373ce68/octoprint_tplinksmartplug/__init__.py#L500-L524

shanejimmink, 2021-06-22:
Thanks. I am perplexed why it is reporting 135 W / 1.05A. There doesn't seem to be a setting in the KASA app for voltage (you would think it would know). I don't think it is a power factor issue (maybe?). Is this seen by anyone else?