jneilliii/OctoPrint-TPLinkSmartplug #274: [BUG]:

NVDUU, 2021-06-24:
**Describe the bug**
When I have the option for off with shutdown enabled the smart plug that is setup for my printer does not turn off after shutting down the pi through octoprint, however the octoprint button still does work to manually turn it off.

**Expected behavior**
If I understand the feature correctly it should turn off the smart plug after the pi is shutdown when this option is enabled

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - OctoPrint Version 1.6.1
 - Plugin Version 1.0.1
 - Printer: Ender 3 V2
 - Pi model: Raspberry Pi 3 model B+

**Additional context**
Using the TP-Link Kasa KP105 model with hardware version 1.0 and firmware version 1.0.6

jneilliii, 2021-06-24:
Enable debug logging, restart octoprint and try again. If our is still not acting as expected share screenshots of all your settings and the octoprint and plugin_tplinksmartplug_debug.log file here in a comment.

NVDUU, 2021-06-25:
Hi have gone through the steps you have mentioned and it is still not turning off the plug after I click shutdown system in octoprint. Attached are the log file and screenshots, thank you for any help that you can give!

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6714650/plugin_tplinksmartplug_debug.log)

[TP-link settings screenshots.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6714735/TP-link.settings.screenshots.zip)

jneilliii, 2021-06-25:
Yeah, this is strange. Seems like it's trying to do it but the plug isn't responding.

```
[2021-06-25 08:59:39,015] DEBUG: powering off 192.168.1.74 due to shutdown event.
[2021-06-25 08:59:39,019] DEBUG: Turning off 192.168.1.74.
[2021-06-25 08:59:39,020] INFO: Turning off 192.168.1.74 at 2021-06-25 08:59:39.019775
[2021-06-25 08:59:39,022] DEBUG: {'autoConnect': True, 'autoConnectDelay': '5', 'autoDisconnect': True, 'autoDisconnectDelay': '15', 'automaticShutdownEnabled': False, 'btnColor': '#808080', 'countdownOffDelay': '10', 'countdownOnDelay': 1, 'currentState': 'on', 'displayWarning': True, 'emeter': {'get_realtime': {}}, 'event_on_disconnect': False, 'event_on_error': False, 'event_on_shutdown': True, 'event_on_startup': True, 'event_on_upload': False, 'gcodeCmdOff': True, 'gcodeCmdOn': False, 'gcodeEnabled': False, 'gcodeOffDelay': '30', 'gcodeOnDelay': 0, 'gcodeRunCmdOff': 'G28 R10 X Y ', 'gcodeRunCmdOn': '', 'icon': 'icon-bolt', 'ip': '192.168.1.74', 'label': 'Printer Power', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': True, 'useCountdownRules': False, 'warnPrinting': True}
[2021-06-25 08:59:39,023] DEBUG: sending gcode commands to printer.
[2021-06-25 08:59:54,058] DEBUG: IP 192.168.1.74 is valid.
[2021-06-25 08:59:54,059] DEBUG: Sending command {'system': {'set_relay_state': {'state': 0}}} to 192.168.1.74
[2021-06-25 08:59:54,060] DEBUG: Could not connect to 192.168.1.74.
[2021-06-25 08:59:54,061] DEBUG: None
```

I'll try to do some testing with these settings over the weekend and see if I can reproduce the issue or not.

jneilliii, 2021-06-26:
I thought I remembered this error. @NVDUU can you see if running a Pi OS update resolves the issue? It seemed to fix it for this user.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/263#issuecomment-819685613

NVDUU, 2021-06-26:
I did the update as mentioned in that comment however it still isn't turning off as it is supposed to.

Here's an updated log file if needed: 
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/6720007/plugin_tplinksmartplug_debug.log)

jneilliii, 2021-06-26:
That's unfortunate. It appears to still have the same error.

```
[2021-06-26 12:22:05,183] DEBUG: powering off 192.168.1.74 due to shutdown event.
[2021-06-26 12:22:05,183] DEBUG: Turning off 192.168.1.74.
[2021-06-26 12:22:05,184] INFO: Turning off 192.168.1.74 at 2021-06-26 12:22:05.183915
[2021-06-26 12:22:05,185] DEBUG: {'autoConnect': True, 'autoConnectDelay': '5', 'autoDisconnect': True, 'autoDisconnectDelay': '15', 'automaticShutdownEnabled': False, 'btnColor': '#808080', 'countdownOffDelay': '30', 'countdownOnDelay': 1, 'currentState': 'on', 'displayWarning': True, 'emeter': {'get_realtime': {}}, 'event_on_disconnect': False, 'event_on_error': False, 'event_on_shutdown': True, 'event_on_startup': True, 'event_on_upload': False, 'gcodeCmdOff': True, 'gcodeCmdOn': False, 'gcodeEnabled': False, 'gcodeOffDelay': '30', 'gcodeOnDelay': 0, 'gcodeRunCmdOff': 'G28 R10 X Y ', 'gcodeRunCmdOn': '', 'icon': 'icon-bolt', 'ip': '192.168.1.74', 'label': 'Printer Power', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': True, 'useCountdownRules': False, 'warnPrinting': True}
[2021-06-26 12:22:05,186] DEBUG: sending gcode commands to printer.
[2021-06-26 12:22:20,304] DEBUG: IP 192.168.1.74 is valid.
[2021-06-26 12:22:20,305] DEBUG: Sending command {'system': {'set_relay_state': {'state': 0}}} to 192.168.1.74
[2021-06-26 12:22:20,306] DEBUG: Could not connect to 192.168.1.74.
[2021-06-26 12:22:20,307] DEBUG: None
```

I've been printing this weekend so haven't had a chance to debug yet. Thanks for trying the updates.

NVDUU, 2021-06-28:
I don't know if it helps but when I press reboot system through octoprint the plug does actually turn off then turn on again as the pi reboots, but pressing shutdown still doesn't work.

jneilliii, 2021-07-08:
What other plugin's do you have installed? Might be easiest to get a list from octoprint.log from during the bootup process. You can grab the section that lists all the plugins and associated versions.

NVDUU, 2021-07-16:
Hi sorry for the late reply I kinda forgot to check github, here are the plugins I have installed:

|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Autoscroll (0.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_autoscroll
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
|  BetterHeaterTimeout (1.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BetterHeaterTimeout
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BLTouch
|  Camera Settings (0.2.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_CameraSettings
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Detailed Progress (0.2.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Exclude Region (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_excluderegion
|  Fan Speed Control (0.2.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_fanspeedslider
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.5.31) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Floating Navbar (0.3.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_floatingnavbar
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  M73 Progress Plugin (0.2.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_m73progress
|  Navbar Temperature Plugin (0.14) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_navbartemp
|  Ngrok Tunnel (0.4.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ngrok
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.6.14) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrettyGCode (1.2.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prettygcode
|  PrintTimeGenius Plugin (2.2.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Simple Emergency Stop (1.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_simpleemergencystop
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Tab Order (0.5.12) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_taborder
|  Themeify (1.2.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_themeify
|  TouchUI (0.3.18) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_touchui
|  TP-Link Smartplug (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WebcamTab (0.2.11) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webcamtab

jneilliii, 2021-07-18:
Yeah, consensus on Discord is that none of these seem like they could block shutdown events. I'm wondering if the network stack is getting shutdown prior to octoprint's service during a standard shutdown of the pi. Is your shutdown system command `sudo shutdown` or something else? I may look into the possibility of an on_before_shutdown event, but IIRC that doesn't exist.

NVDUU, 2021-07-19:
[image]

These are the shutdown and restart settings being used, they are the default ones that came with octoprint I haven't touched them.

jneilliii, 2021-07-19:
do me a favor and try taking the `-h now` part off and see if that changes anything.

jneilliii, 2021-07-19:
actually, further discussion on Discord believes this won't make a difference.

NVDUU, 2021-07-22:
Yeah I tried removing the -h now part and still no luck

mattfox27, 2021-07-22:
Im having the same issue...Works via lightning bolt in header but does not turn off after print is complete

jneilliii, 2021-07-22:
Yeah, I'm pretty sure this boils down to what I suspect. The network is shutting down prior to OctoPrint completing its shutdown process. Unfortunately, there is no before_shutdown event that I could respond to. 

The only workaround that I could make to you guys at this time is possibly going the other route using idle timers (or gcode triggers), countdown timers and system command (to do the pi shutdown command) options. Because the lines are showing up in the log I know the commands are getting processed, I just think they aren't going anywhere.

jneilliii, 2021-07-22:
> Im having the same issue...Works via lightning bolt in header but does not turn off after print is complete

this sounds like a different issue, did you enable any of the settings for automatically powering off after print complete via gcode or idle timers?

mattfox27, 2021-07-22:
I believe i did set it on timers...what's the best way to set it?

mattfox27, 2021-07-22:
Would I just do like M81 ip of tp link

jneilliii, 2021-07-22:
I use idle timers personally and have no issues with it powering off my plugs after the set conditions. Here is my settings...

[image]

[image]

my best guess is you didn't check the idle option in the individual plug's settings.

mattfox27, 2021-07-22:
Awesome, thanks!

jneilliii, 2021-07-23:
We had an idea of changing the service configuration, but that's not going to work out after further testing. I am able to reproduce the issue even after performing the following steps...

```
sudo nano /etc/systemd/system/octoprint.service
```
make it look like this.
```
[Unit]
Description=The snappy web interface for your 3D printer
Requires=network-online.target

[Service]
Environment="HOST=127.0.0.1"
Environment="PORT=5000"
Type=simple
User=pi
ExecStart=/home/pi/oprint/bin/octoprint serve --host=${HOST} --port=${PORT}

[Install]
WantedBy=multi-user.target
```
then press ctrl+x and answer yes to save the file and then hit enter to keep the same filename. then run
```
sudo systemctl daemon-reload
sudo service octoprint restart
```

but this does seem to be related to an issue posted to core octoprint. I have posted an update on that issue to see if there is a potential solution or not.

mattfox27, 2021-07-23:
So does that mean once it goes below 50deg for 15 min it will turn off?

jneilliii, 2021-07-23:
@mattfox27 that means after being idle for 15 minutes (no print activity) it will power off, if the temperature is above 50 it will wait until it is below 50 and then power off

mattfox27, 2021-07-23:
It worked...thank you so much for your help

jneilliii, 2021-07-24:
@NVDUU I think I was able to work around this issue in the above commit. I've released version 1.02.rc1 for you to test. Change the release channel for the plugin in Software Update settings to `Release Candidate` and install the update when prompted. My testing proved positive.

NVDUU, 2021-07-28:
Hi sorry for the late reply as I was busy with school, I tried the new release and followed the steps you mentioned, however the off with shutdown is still not working.

jneilliii, 2021-07-28:
Interesting, it was working as expected for me after the update. Could you share screenshots of your settings again or have they not changed from the initial post?

NVDUU, 2021-07-28:
They have not changed from the initial post, all I did was update the plugin to the new release.

jneilliii, 2022-03-21:
How are you initiating the power off @ssombra, with idle timeout or gcode triggers?

jneilliii, 2022-03-21:
Oh, I got you. For that you need top use the timer options because what happens is octoprint shots the pi down before the command can be sent ri the printer I suspect.

ssombra, 2022-03-21:
You mean "Timer off delay" under the plug settings? but wouldn't that have the same outcome of octoprint shutting the pi down?

jneilliii, 2022-03-21:
It may or may not still have the issue, but I know that will be the only way it can work if the pi goes off. It sends the command to the plug to use its own countdown timer to power off at a later (configured) time.

ssombra, 2022-03-23:
I tried using the timer and still doesn't work :(

jneilliii, 2022-03-23:
enable debug logging and restart octoprint. share screenshots of your settings and share the plugin_tplinksmartplug_debug.log file here in a comment.

ssombra, 2022-03-27:
Sure. 

This is the debug file conent:

```
[2022-03-27 20:46:02,850] DEBUG: abortTimeout: 30
[2022-03-27 20:46:02,855] DEBUG: powerOffWhenIdle: False
[2022-03-27 20:46:02,861] DEBUG: idleTimeout: 30
[2022-03-27 20:46:02,867] DEBUG: idleIgnoreCommands: M105
[2022-03-27 20:46:02,873] DEBUG: idleTimeoutWaitTemp: 50
[2022-03-27 20:46:02,879] DEBUG: powering on due to startup.
[2022-03-27 20:46:26,946] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.4.44/1')])
[2022-03-27 20:46:26,947] DEBUG: Checking status of 192.168.4.44/1.
[2022-03-27 20:46:26,947] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-03-27 20:46:26,948] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:46:26,949] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-03-27 20:46:26,950] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:46:26,950] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.4.44
[2022-03-27 20:46:26,977] DEBUG: « U"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 201015 Rel.173920","hw_ver":"2.0","model":"KP303(US)","deviceId":"8006671C83AA712BE3223471C7C777C91ED7BC67","oemId":"3DB05524283BBF3A7D248CBB9389029D","hwId":"9FD90762FBE6CD307B19E70B5EE345E2","rssi":-72,"latitude_i":192732,"longitude_i":-991508,"alias":"TP-LINK_Power Strip_A728","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"C0:06:C3:AB:A7:28","updating":0,"led_off":0,"children":[{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6700","state":1,"alias":"AM8","on_time":14,"next_action":{"type":-1}},{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6701","state":1,"alias":"RPi","on_time":762,"next_action":{"type":-1}},{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6702","state":1,"alias":"Enchufe 3","on_time":3804,"next_action":{"type":-1}}],"child_num":3,"ntc_state":0,"err_code":0}}}
[2022-03-27 20:46:26,981] DEBUG: get_device_id response: 8006671C83AA712BE3223471C7C777C91ED7BC6700
[2022-03-27 20:46:26,981] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': ['8006671C83AA712BE3223471C7C777C91ED7BC6700']}} to 192.168.4.44
[2022-03-27 20:46:27,092] DEBUG: « U"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 201015 Rel.173920","hw_ver":"2.0","model":"KP303(US)","deviceId":"8006671C83AA712BE3223471C7C777C91ED7BC67","oemId":"3DB05524283BBF3A7D248CBB9389029D","hwId":"9FD90762FBE6CD307B19E70B5EE345E2","rssi":-72,"latitude_i":192732,"longitude_i":-991508,"alias":"TP-LINK_Power Strip_A728","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"C0:06:C3:AB:A7:28","updating":0,"led_off":0,"children":[{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6700","state":1,"alias":"AM8","on_time":14,"next_action":{"type":-1}},{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6701","state":1,"alias":"RPi","on_time":762,"next_action":{"type":-1}},{"id":"8006671C83AA712BE3223471C7C777C91ED7BC6702","state":1,"alias":"Enchufe 3","on_time":3804,"next_action":{"type":-1}}],"child_num":3,"ntc_state":0,"err_code":0}}}
[2022-03-27 20:46:27,095] DEBUG: TIM
[2022-03-27 20:47:17,065] DEBUG: powering off 192.168.4.44/1 due to shutdown event.
[2022-03-27 20:47:17,066] DEBUG: Turning off 192.168.4.44/1.
[2022-03-27 20:47:17,067] INFO: Turning off 192.168.4.44/1 at 2022-03-27 20:47:17.066539
[2022-03-27 20:47:17,069] DEBUG: {'autoConnect': True, 'autoConnectDelay': 10, 'autoDisconnect': True, 'autoDisconnectDelay': 0, 'automaticShutdownEnabled': False, 'btnColor': '#808080', 'countdownOffDelay': '10', 'countdownOnDelay': 1, 'currentState': 'unknown', 'displayWarning': True, 'emeter': {'get_realtime': {}}, 'event_on_disconnect': False, 'event_on_error': False, 'event_on_shutdown': True, 'event_on_startup': False, 'event_on_upload': False, 'gcodeCmdOff': False, 'gcodeCmdOn': False, 'gcodeEnabled': False, 'gcodeOffDelay': 0, 'gcodeOnDelay': 0, 'gcodeRunCmdOff': '', 'gcodeRunCmdOn': '', 'icon': 'fas fa-print', 'ip': '192.168.4.44/1', 'label': 'AM8', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': False, 'useCountdownRules': True, 'warnPrinting': False}
[2022-03-27 20:47:17,072] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:47:17,073] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-03-27 20:47:17,076] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:47:17,077] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.4.44
[2022-03-27 20:47:17,080] DEBUG: Could not connect to 192.168.4.44.
[2022-03-27 20:47:17,081] DEBUG: get_device_id response: False
[2022-03-27 20:47:17,081] DEBUG: Sending command {'count_down': {'delete_all_rules': None}, 'context': {'child_ids': [False]}} to 192.168.4.44
[2022-03-27 20:47:17,082] DEBUG: Could not connect to 192.168.4.44.
[2022-03-27 20:47:17,082] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:47:17,084] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-03-27 20:47:17,084] DEBUG: IP 192.168.4.44 is valid.
[2022-03-27 20:47:17,084] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.4.44
[2022-03-27 20:47:17,085] DEBUG: Could not connect to 192.168.4.44.
[2022-03-27 20:47:17,086] DEBUG: get_device_id response: False
[2022-03-27 20:47:17,086] DEBUG: Sending command {'count_down': {'add_rule': {'enable': 1, 'delay': 10, 'act': 0, 'name': 'turn off'}}, 'context': {'child_ids': [False]}} to 192.168.4.44
[2022-03-27 20:47:17,086] DEBUG: Could not connect to 192.168.4.44.
[2022-03-27 20:47:17,088] DEBUG: None
[2022-03-27 20:47:40,748] DEBUG: abortTimeout: 30
[2022-03-27 20:47:40,751] DEBUG: powerOffWhenIdle: False
[2022-03-27 20:47:40,754] DEBUG: idleTimeout: 30
[2022-03-27 20:47:40,755] DEBUG: idleIgnoreCommands: M105
[2022-03-27 20:47:40,755] DEBUG: idleTimeoutWaitTemp: 50
[2022-03-27 20:47:40,756] DEBUG: powering on due to startup.
```
These are my settings:
PLUG:

[image]

General:
[image]

When I do a Octoprint restart, the plug does go off, but it doesn't when a Shutdown is performed.

jneilliii, 2022-03-30:
Thanks, this helps a little. I notice in that log it is getting the shutdown sequence initiating and trying to send to the plug, but the plug isn't responding. I notice you're using a multi-plug device based on the IP, which is what I also use so I should be able to debug this locally. My best guess is that the network is getting powered down prior to OctoPrint's shutdown sequence completing, so may need to move the shutdown into a different spot in the code, possibly intercepting the API call to shutdown instead of using the shutdown event so it happens prior to the actual command being run on the pi.

ssombra, 2022-03-30:
Yes, that is correct, it is a multi-plug, I think it is a the KP303. Thanks for taking the time to look at this, please let me know if you need something else from me.

jneilliii, 2022-03-30:
Not sure when I'll get around to debugging, but if your end goal is shutting down the pi I could recommend a safe shutdown procedure using the settings documented in my other power plugin. The layout of the settings is a little different, but they are roughly the same settings and uses the idle timer to send the power off command to the plug and then runs the shutdown command. 

https://github.com/jneilliii/OctoPrint-Tasmota#examples

ssombra, 2022-03-30:
I will take a look, thanks

moccor, 2022-04-02:
Is this still a problem? I tried so many different things, but I still can't get devices to shut down when I tell Octoprint to shutdown.

I think the problem is like you mentioned to others, that the shutdown happens before the plugin starts the timer commands. I can see this happen in real time, because if you open the Kasa app on your smartphone and do a restart, the timers will pop up for the devices no problem. If you do a shutdown however, the timers don't show up for me and the plugs do not turn off.

Would a very easy workaround for this be to simply add a button in the "power" dropdown (or a button anywhere at all) that is specially for executing your TP-Link Editor Timers and Then "sudo shutdown now"?

jneilliii, 2022-04-03:
Yeah, that's kind of how the idle example linked above would work. I have an idea of how to circumvent the issue.

moccor, 2022-04-04:
Yeah that's understandable. I do feel like many people would find a button to be useful as well though in some scenarios. But I am not sure how much of a trouble that is to add. I hope the idea works out, and thanks for the awesome plugin

closed 2022-04-17 by jneilliii

borkedporcupine, 2023-05-28:
This  thread is pretty old, not sure if it's being monitored.  I ended up here for this same issue.  As a potential solution what about presenting a shutdown option in the bar in octopi like the lightning bolt but for "system shutdown" so you could issue the plug to power down then issue the shutdown command to octopi to shut it down.  It would keep the order of operations in the correct sequence.

PR #304 (2022-04-17): 1.0.2
* resolves issue related to shutdown monitoring not powering off devices due to SIGTERM, resolves #274
* fix timelapse race condition, resolves for #281
* add JSON response to on/off API commands
* add `Always on Upload` option to not require start print from slicer upload, resolves #190, resolves #267
* replace deprecated getBoolen with get_boolean
* remove duplicated bundling of fontawesome 5 if running on OctoPrint 1.5.0+