jneilliii/OctoPrint-TPLinkSmartplug #275: [Feature Request]: delay between connecting and starting print (after powering on)

kml-pt, 2021-06-29:
Hi there,
My printer takes some time to initialize after OctoPrint establishes the serial connection. Would you be willing to add an optional, configurable delay between when the serial connection is established, and when the print starts? I looked for a way to do this in OctoPrint's core settings, and in this plugin's settings, but could not find a way to do it. It's only an issue after the sequence of upload / auto-on / auto-connect / start print. After the auto-connect, the printer isn't quite ready to start a print with the current timing.

Thank you for the time and effort you've donated to the OctoPrint community.

jneilliii, 2021-06-29:
The delay in seconds underneath the Auto connect option is the delay between powering on the printer and establishing the connection. What you are asking for is a delay on the other side of that connection? The plugin reacts to the printer connected event to start the uploaded file, so usually that means all the handshakes have completed and the printer is ready to go. You might be able to workaround this with a dwell command (G4 S510) in your `After connection to printer is established` GCODE Scripts section of OctoPrint's settings.

kml-pt, 2021-06-29:
That worked wonderfully, thank you for the idea!

closed 2021-06-29 by kml-pt