jneilliii/OctoPrint-TPLinkSmartplug #279: TP-Link SmartPlug - lights come on too late for snapshot

Andyinparadise, 2021-08-03:
I have set up the plugin utilising the @TPLINKON and @TPLINKOFF and it is working fine. The lights are coming on too late for the snapshot. I can set the lights to stay on for 3 or 4 seconds but how can I get them to come on any earlier?
Is there something I am missing or is this a limitation of the TP-Link hardware/software?

Thanks

jneilliii, 2021-08-03:
I guess it depends on the timing/speed of movements and snapshots being taken. The only thing I can think of that might work is adding in a delay after the on command prior to the snapshot being taken. You should be able to do this with the dwell command [G4](https://reprap.org/wiki/G-code#G4:_Dwell) but I don't know if that will resolve this or not.

Andyinparadise, 2021-08-04:
Fantastic, I added G4 S1 and it seems to be working ok now.
Thanks

Andyinparadise, 2021-08-04:
Donation made, thanks again.

jneilliii, 2021-08-04:
appreciate the support

closed 2021-08-04 by jneilliii