jneilliii/OctoPrint-TPLinkSmartplug #281: [BUG]: Power off during print when timelapse finishes

luiselse, 2021-08-17:
**Describe the bug**
When a long print finishes and the timelapse takes long time to process, if a new print starts, when the timelapse finishes the printer will be turned off, despite the fact that it's printing.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a long print (so that the timelapse will take a long time to finish)
2. Once the print finishes start a new print before the timelapse finishes
3. During the new print, once the timelapse finishes, the printer will be powered of

**Expected behavior**
The printer should never be powered of if it's actively printing.

**Versions:**
 - OctoPrint 1.6.1
 - TP-Link Smartplug (1.0.1)

**Logs**
[2021-08-16 19:23:23,178] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:28,182] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:33,187] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:38,188] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:43,194] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:48,198] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:53,200] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:23:58,203] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:03,209] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:08,215] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:13,218] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:18,224] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:23,229] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:28,232] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:33,234] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:38,238] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:43,243] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:48,244] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:53,250] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:24:58,256] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:03,261] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:05,571] DEBUG: Checking status of 192.168.1.40.
[2021-08-16 19:25:05,572] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-08-16 19:25:05,572] DEBUG: IP 192.168.1.40 is valid.
[2021-08-16 19:25:05,572] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.40
[2021-08-16 19:25:05,662] DEBUG: « b°"system":{"get_sysinfo":{"sw_ver":"1.0.16 Build 210205 Rel.163735","hw_ver":"1.0","model":"KP115(UK)","deviceId":"8006173E89958940D70EA00B80E85FEC1DDF0D10","oemId":"C7A36E0C2D4BAB44DED6EF0870AC707F","hwId":"39E8408ED974DD69D8A77D9F8781637E","rssi":-29,"latitude_i":515894,"longitude_i":-3745,"alias":"Printer","status":"new","obd_src":"tplink","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"90:9A:4A:07:4A:3F","updating":0,"led_off":0,"relay_state":1,"on_time":111132,"icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","active_mode":"none","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2021-08-16 19:25:05,704] DEBUG: TIM:ENE
[2021-08-16 19:25:05,705] DEBUG: IP 192.168.1.40 is valid.
[2021-08-16 19:25:05,705] DEBUG: Sending command {'emeter': {'get_realtime': {}}} to 192.168.1.40
[2021-08-16 19:25:05,749] DEBUG: «  r¢"emeter":{"get_realtime":{"current_ma":1077,"voltage_mv":247731,"power_mw":139463,"total_wh":8663,"err_code":0}}}
[2021-08-16 19:25:08,263] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:13,268] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:18,272] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:20,263] DEBUG: Checking statuses during print progress (20).
[2021-08-16 19:25:20,318] DEBUG: Resetting idle timer during print progress (20)...
[2021-08-16 19:25:21,292] DEBUG: Checking status of 192.168.1.40.
[2021-08-16 19:25:21,292] DEBUG: {'system': {'get_sysinfo': {}}}
[2021-08-16 19:25:21,292] DEBUG: IP 192.168.1.40 is valid.
[2021-08-16 19:25:21,293] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.40
[2021-08-16 19:25:21,359] DEBUG: « b°"system":{"get_sysinfo":{"sw_ver":"1.0.16 Build 210205 Rel.163735","hw_ver":"1.0","model":"KP115(UK)","deviceId":"8006173E89958940D70EA00B80E85FEC1DDF0D10","oemId":"C7A36E0C2D4BAB44DED6EF0870AC707F","hwId":"39E8408ED974DD69D8A77D9F8781637E","rssi":-28,"latitude_i":515894,"longitude_i":-3745,"alias":"Printer","status":"new","obd_src":"tplink","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"90:9A:4A:07:4A:3F","updating":0,"led_off":0,"relay_state":1,"on_time":111148,"icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","active_mode":"none","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2021-08-16 19:25:21,401] DEBUG: TIM:ENE
[2021-08-16 19:25:21,402] DEBUG: IP 192.168.1.40 is valid.
[2021-08-16 19:25:21,402] DEBUG: Sending command {'emeter': {'get_realtime': {}}} to 192.168.1.40
[2021-08-16 19:25:21,444] DEBUG: «  r¢"emeter":{"get_realtime":{"current_ma":1034,"voltage_mv":248305,"power_mw":131912,"total_wh":8663,"err_code":0}}}
[2021-08-16 19:25:23,280] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:28,281] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:33,287] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:38,289] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:43,293] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:48,300] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:53,302] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:25:58,303] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:03,304] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:08,308] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:13,316] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:18,321] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:23,324] DEBUG: Waiting for timelapse before shutting off power...
[2021-08-16 19:26:24,247] DEBUG: Timelapse generation finished: drawer_organizer_mine_0.16mm_PLA_ENDER5_1d0h55m_20210815123313.mp4. Return Code: completed
[2021-08-16 19:26:28,384] DEBUG: Starting abort power off timer.
[2021-08-16 19:27:28,506] DEBUG: Automatically powering off enabled plugs.
[2021-08-16 19:27:28,507] DEBUG: Turning off 192.168.1.40.
[2021-08-16 19:27:28,508] INFO: Turning off 192.168.1.40 at 2021-08-16 19:27:28.507931
[2021-08-16 19:27:28,510] DEBUG: {'autoConnect': True, 'gcodeRunCmdOn': '', 'gcodeCmdOn': False, 'ip': '192.168.1.40', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'event_on_shutdown': False, 'emeter': {'get_realtime': {'total_wh': 5090, 'current_ma': 0, 'power_mw': 0, 'voltage_mv': 245635, 'err_code': 0}}, 'countdownOnDelay': 1, 'label': 'Printer', 'warnPrinting': True, 'thermal_runaway': False, 'gcodeCmdOff': False, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': 1, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'event_on_upload': True, 'useCountdownRules': False, 'event_on_startup': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'sysCmdOn': False, 'gcodeRunCmdOff': '', 'currentState': 'on', 'sysCmdOffDelay': 0}
[2021-08-16 19:27:28,629] DEBUG: IP 192.168.1.40 is valid.
[2021-08-16 19:27:28,630] DEBUG: Sending command {'system': {'set_relay_state': {'state': 0}}} to 192.168.1.40
[2021-08-16 19:27:28,716] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}

jneilliii, 2021-08-17:
Thanks for your report. This is a combination that I never thought to look out for since I don't personally do timelapses, just know that I have accounted for timelapse to be running when the off command from idle timeout was sent. That is probably what is happening here. I need to add some logic back in to that idle timeout once triggered to back out of any of the other checks (temp level, timelapse generation, etc.) if a print is started or the printer is in a printing state.

luiselse, 2021-08-20:
Thanks for the really fast response, I was hesitant if I could answer with any valuable comment, I don't want to write something that doesn't add any value.

It's not a great help but after turning the timelapse off this problem no longer happens, so it's clearly what I wrote in the original post, when the timelapse finishes, this plug-in turns the printer off even if the printer is printing.

For now I'll just leave the timelapse off.

Thank you.

jneilliii, 2021-08-21:
I just released version 1.0.2rc2 that I think will resolve the issue for you. Unfortunately, I could not reproduce in my dev environment on my laptop because the timelapse finishes so quickly the idle timer doesn't have enough time to kick in before it's done rendering the timelapse. If you're willing to give it a try and potentially have a failed print you can change the plugin's release channel in OctoPrint's software update settings to Release Candidate and install the update when prompted.

closed 2022-04-23 by jneilliii