jneilliii/OctoPrint-TPLinkSmartplug #284: [Feature Request]: Remove data points when printer is off

GrimWreeper, 2021-09-07:
**Is your feature request related to a problem? Please describe.**
This is a request. I think it would be useful to drop (either delete or not log) the datapoints gathered while the printer is off.
Maybe it's just me but I'm just super interested in the amount of power being used when I know the printer is off

**Describe the solution you'd like**
Ideally this would be user configured so as to not break any existing workflows for folks.
 
Thank you for this plugin

jneilliii, 2021-09-07:
Will look into adding an option for this in the future.

puterboy, 2024-02-09:
I submitted a PR that fixes this https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/pull/352

jneilliii, 2025-08-17:
Has been available as 1.0.4rc1 for some time, and recently released 1.0.4 official. Hopefully, this doesn't get broken in the latest refactored version 2.0.0rc1.

closed 2025-08-17 by jneilliii