jneilliii/OctoPrint-TPLinkSmartplug #286: [BUG]: Using the plugin freezes the feed from a USB Webcam

NuarStanger, 2021-09-30:
I had issues with my web cam not working after some new updates went through. I decided to ignore the issue for quite a few months... until I got frustrated and decided to reinstal octoprint from scratch on a brand new MicroSD card.
Everything was working well.
I installed a few plugins (Themeify, BLTouch, Bed Visualizer and finaly the TPLinkSmartplug)
After a reboot of the Pi, the camera still worked, then I tried to preheat the printer and noticed the feed is frozen, I refreshed the page and the webcam feed didn't load.
I restarted the entire Pi.
Web cam worked again, then I clicked the tplinksmartplug icon to power the printer on and noticed the feed freezes as soon as I do that.
This happens every single time.
Reboot the Pi, webcam works, I click the tplinksmartplug icon to turn the printer on and the feed freezes right away.
The pi is powered off it's own power supply, not from the printer (not sure if that needed to be mentioned) and the power supply has been changed to a new one as part of trying to troubleshoot this problem.
This is on a Raspberry Pi 3B+
All this started happening after an octoprint update a few months ago, newer updates did not fix the issue.

jneilliii, 2021-09-30:
This plugin does nothing to the camera feed, mjpgstreamer, or anything else on the USB side of things and only communicates over the network to your device. I use this plugin regularly with TPLink devices and have never experienced the issue you are describing with the camera feed on any of my instances. 

Could you please share your plugin's settings so I can have a better understanding of your setup and possibly be able to reproduce the issue. Also, enabling debug logging for the plugin and providing your octoprint.log and plugin_tplinksmartplug_debug.log files after enabling would be beneficial in diagnosing your situation.

NuarStanger, 2021-09-30:
I don't see how the plugin could mess with the webcam, my guess is that it must be something in the actual octoprint software that got changed that is causing this issue.
I will try to get all that info for you within the next day or two. 
(Posted this while on my lunch break at work lol)
Thank you for such a quick response!

NuarStanger, 2021-10-01:
Do you have an email I could send the two logs to?

As for the plugin settings, I only have a warning prompt, auto connect and auto disconnect checked. The default values for now since it's a fresh octoprint instal and I was just getting it all setup.

jneilliii, 2021-10-01:
Yeah, you can email me at the address on the plugin page or alternatively, you can just drag them into a comment here on github to upload direct.

jneilliii, 2021-10-01:
Ok, so I don't see anything strange in those logs at all. Would it be possible for you to check your browser's developer tools, on the console tab to see if there are any errors when this is happening? Typically, that is accessed by pressing F12 on the keyboard. You would probably see something like 404 file not found or something else similar. Another thing to check to see if it's the OctoPrint interface or an incorrect stream URL. Try opening the stream directly like http://octopi.local/webcam/?action=stream to see if it's freezing there too. And the last thing to try is opening OctoPrint in an incognito window to disable any possible browser extensions that may be interfering.

NuarStanger, 2021-10-01:
This happens regardless of browser. I can do it on my laptop, my pc, my
phone or a tablet.
I really feel the issue is in the actual octoprint after one of the
updates. Something there doesn't play well with the plugin and the webcam
or usb connections.
I will try to instal octoprint on a raspberry pi 4 and see if it happens
there too. Just out of curiosity.

jneilliii, 2021-10-04:
Just saw this posted on the forum and sounds similar to your situation, but I don't think they have this plugin installed. 

https://community.octoprint.org/t/intermittent-issue-in-which-rpi-hq-camera-feed-freezes/37627?u=jneilliii

NuarStanger, 2021-10-04:
I took a short video.
Laptop is to the left of the door, door in the center and printer to the
right of the door.

https://youtu.be/IN2YYbdUt-M

NuarStanger, 2021-10-04:
I have a rpi 4 that i'm going to try. I'll see if this issue is pi3b+
specific.
After that i'll try a raspberry pi camera with a ribbon cable on the 3b+

jneilliii, 2021-10-04:
That is so strange.

NuarStanger, 2021-10-04:
I agree. It makes no friggin sense. I've been using your plugin for some
time now. On my old cr10s and now this ener 3 pro. I love it... one of the
recent octoprint updates broke it. I'll try the pi cam and see if it works
with that since it's not USB.

jneilliii, 2021-10-04:
curious to see what `journalctl -u webcamd` reports at the time of the freeze.

NuarStanger, 2021-10-04:
Not sure what you mean.
My Linux knowledge is very limited.
You want me to ssh into the pi and run the command after the webcam freezes?

jneilliii, 2021-10-04:
Correct

NuarStanger, 2021-10-05:
***@***.***:~ $ journalctl -u webcamd
-- Logs begin at Sun 2021-10-03 10:35:47 EDT, end at Mon 2021-10-04
21:21:50 EDT
Oct 03 10:35:51 octopi systemd[1]: Starting the OctoPi webcam daemon with
the us
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
applica
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: MJPG
Streamer Ve
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Using V4L2
devic
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Desired
Resoluti
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Frames Per
Secon
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
Format..........
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
TV-Norm.........
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
www-folder-path.
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: HTTP TCP
port...
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: HTTP Listen
Addr
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
username:passwor
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
commands........
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
input p
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
output
Oct 03 10:35:53 octopi systemd[1]: Started the OctoPi webcam daemon with
the use
Oct 04 21:21:46 octopi mjpg_streamer[467]: MJPG-streamer [467]: Error
grabbing f
Oct 04 21:21:46 octopi mjpg_streamer[467]: MJPG-streamer [467]: cleaning up
reso
lines 1-19/19 (END)...skipping...
-- Logs begin at Sun 2021-10-03 10:35:47 EDT, end at Mon 2021-10-04
21:21:50 EDT. --
Oct 03 10:35:51 octopi systemd[1]: Starting the OctoPi webcam daemon with
the user specified config...
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
application
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: MJPG
Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Using V4L2
device.: /dev/video0
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Desired
Resolution: 640 x 480
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: Frames Per
Second.: -1
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
Format............: JPEG
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
TV-Norm...........: DEFAULT
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
www-folder-path......: ./www-octopi/
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: HTTP TCP
port........: 8080
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: HTTP Listen
Address..: 127.0.0.1
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
username:password....: disabled
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]:
commands.............: disabled
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
input plugin input_uvc.so
Oct 03 10:35:52 octopi mjpg_streamer[467]: MJPG-streamer [467]: starting
output plugin: output_http.so (ID: 00)
Oct 03 10:35:53 octopi systemd[1]: Started the OctoPi webcam daemon with
the user specified config.
Oct 04 21:21:46 octopi mjpg_streamer[467]: MJPG-streamer [467]: Error
grabbing frames
Oct 04 21:21:46 octopi mjpg_streamer[467]: MJPG-streamer [467]: cleaning up
resources allocated by input thread
~

jneilliii, 2021-10-05:
ah, it got clipped, can you make your window wider so the text isn't getting clipped off? also if you can check the time and see if you can identify it coordinating with anything in the log.

NuarStanger, 2021-10-05:
I'll try again tonight.

NuarStanger, 2021-10-08:
***@***.***:~ $ journalctl -u webcamd
-- Logs begin at Tue 2021-10-05 22:20:41 EDT, end at Thu 2021-10-07
22:21:32 EDT. --
Oct 05 22:20:45 octopi systemd[1]: Starting the OctoPi webcam daemon with
the user specified config...
Oct 05 22:20:46 octopi root[464]: Starting USB webcam
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: starting
application
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: MJPG
Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: Using V4L2
device.: /dev/video0
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: Desired
Resolution: 640 x 480
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: Frames Per
Second.: -1
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]:
Format............: JPEG
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]:
TV-Norm...........: DEFAULT
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]:
www-folder-path......: ./www-octopi/
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: HTTP TCP
port........: 8080
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: HTTP Listen
Address..: 127.0.0.1
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]:
username:password....: disabled
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]:
commands.............: disabled
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: starting
input plugin input_uvc.so
Oct 05 22:20:46 octopi mjpg_streamer[465]: MJPG-streamer [465]: starting
output plugin: output_http.so (ID: 00)
Oct 05 22:20:47 octopi systemd[1]: Started the OctoPi webcam daemon with
the user specified config.
Oct 07 22:21:32 octopi mjpg_streamer[465]: MJPG-streamer [465]: Error
grabbing frames
Oct 07 22:21:32 octopi mjpg_streamer[465]: MJPG-streamer [465]: cleaning up
resources allocated by input thread
***@***.***:~ $

Sorry for the delay. I had a few really busy days.
I started octoprint, waited a little bit for things to initialize and then
I clicked the icon for the tplinksmartplug plugin to turn the printer on.
As soon as I did that, the webcam feed froze right away and I ran the
command in ssh right away, all within 10 seconds.
What I find very strange is that I have a 2nd smart plug setup that turns
the LEDs on around my printer, I set that up with your plugin as well... I
can use that one all I want and no issues come up. Everything works fine
and webcam is uninterrupted.
I tried switching the plug for the printer power to the other smart plug
that usually runs the led's... when I turned that plug on with the printer
power plugged into it, it crashed the webcam feed just like the first
smartplug.

NuarStanger, 2021-10-09:
> would you say that was about the time that it kinda froze?
> This I find very interesting, and makes me believe that something is
> happening electrically. I wonder if this might be some kind of grounding
> issue, or backpowering
> issue from the printer. If you disconnect the USB cable between the printer
> and pi and power on the printer does the webcam freeze then?
> Something else to try, once it freezes what happens if you restart the
> webcam service by SSH'ing to the pi and running sudo service webcamd
> restart and see if that brings it back or not.

So i tried disconnecting the pi from the printer and started the printer.
The webcam feed did not freeze...
I shut the printer off and plugged the pi back in, turned the printer on,
webcam froze..so then I ran the restart command and it started
working again... so in a way it can be worked around.

The strange thing is, i had the power pin on the usb cable covered with
electrical tape to stop the pi from powering the printer through usb. This
is from when I had the original motherboard in there and then an MKS Gen L.
Now the printer has an SKR 1.3 in there.

I removed the tape from the usb cable and it all acts the same.
Feed freezes when the printer comes online and the restart command gets the
webcam working again.

As for the 1st question, yes that's the exact time the webcam freezes that
the log reports the error.

Sorry for the delayed response, busy week :-/

jneilliii, 2021-10-09:
Yeah, so obviously not a plugin issue. Couple of things I can recommend. One is adding a system menu option to restart the webcam service to make it easier to do that after powering on the printer. SSH to your pi and run `nano /home/pi/.octoprint/config.yaml` and add these settings.

```
system:
  actions:
  - name: Restart Webcam
    action: restartwebcam
    command: sudo service webcamd restart
```
press ctrl+s to save and then ctrl+x to exit.

Then restart OctoPrint and then in that same menu where you restart OctoPrint from you should see a new menu item `Restart Webcam`. 

The second option is using the [gcode system commands](https://plugins.octoprint.org/plugins/gcodesystemcommands/) plugin, and assign a gcode command `OCTO10` to the command `sudo service webcamd restart` and then in OctoPrint's GCODE scripts section add the `OCTO10` command to the `After serial connection to printer is established
` section. That way when the printer powers on, the plugin connects OctoPrint to the printer, and then the webcam service will be restarted automatically. 

And a third option actually is using event subscriptions. It's a little easier in OctoPrint version 1.7.0 (currently in rc3 and will be released soon) because it has an Event Manager built-in (I made it). You would add an event manually in config.yaml (similar to above process to open the file).

```
events:
  subscriptions:
    - event: Connected
      command: "sudo service webcamd restart"
      type: system
```

save and restart OctoPrint. 

Hope one of those options help.