jneilliii/OctoPrint-TPLinkSmartplug #290: [Feature Request]: Group Plugs together

JonSnow88, 2021-12-05:
So I have 1 plug for the printer and 1 plug for the LEDs, what I would like to do is when I power on/off the printer, it will then also power on/off the LEDs automatically without me needing to turn on/off each plug individually. 

Ideally the idle time works for powering off but on, so is there a way to group these plugs? or a system command that can be added to trigger the LED plug?

jneilliii, 2021-12-05:
You should be able to set the settings for both devices the same (minus auto connect/disconnect on the LED) and that would allow both to automatically power on. The other option would be to configure the LEF with gcode options and then send the gcode options on printer connection in OctoPrint's GCODE scripts settings, or via event manager.

JonSnow88, 2021-12-05:
As far as i understand the automatic power on is based on an GCODE file being uploaded to octoprint?

I tend to upload files but sometimes am not ready to print it right away. So for this reason I always turn the printer on manually from the nav bar, what i would like is to automate the LED plug based on the printer plug.

What would be the GCODE command to just power on the LED plug?

My LEDs are hooked up to octoprint and are fully controlled by the PI using another plug in.

jneilliii, 2021-12-05:
> What would be the GCODE command to just power on the LED plug?

M80 192.168.0.2 or @TPLINKON 192.168.0.2, using the plug's ip that's defined in settings. 

> My LEDs are hooked up to octoprint and are fully controlled by the PI using another plug in.

Now I'm confused, why not use that plugin to control it, I bet there a gcode command options for that as well.

JonSnow88, 2021-12-05:
Send: M80 192.168.0.2,
Recv: echo:Unknown command: "M80 192.168.0.2,"

I added this to Run GCODE Commands after ON but it doesn't work, please note I did update the IP address correctly and am not using 192.168.0.2 within my GCODE, same for @TPLINKON.

The reason I want this is so that I can kill power to the LEDs independently from Octoprint, not everyone has access to octoprint  so a manual switch is needed.

jneilliii, 2021-12-05:
and did you enable the gcode option for that specific plug?

JonSnow88, 2021-12-05:
> and did you enable the gcode option for that specific plug?

So on the Printer plug I have GCODE Trigger checked, I also have the plug Home the Y axis right before it turns the printer off and this does work. 

Do I need to turn on GCODE trigger on the LED plug as well?

jneilliii, 2021-12-05:
Cool, closing because I do not want to deal with programming groups of devices.

closed 2021-12-05 by jneilliii