jneilliii/OctoPrint-TPLinkSmartplug #291: [Feature Request]: Support IFTTT via Pushbullet / Pushover messages instead of directly communicating with smart plugs

xxTheGoDxx, 2021-12-12:
So, I am new to 3D printing and looking into making my printer more secure. This plugin is fantastic for that so its even more sad to see that TPLink support is dying due to recent firmware updates.

Here is an alternative:

A lot of smart plugs via their connected cloud services are compatible with IFTTT, among others TPLink Kasa devices, iHome Control SmartPlugs and Meross plugs with likely more to come. 

Ideally of course you could directly hook into IFTTT, but I guess that might be out of the scope of an open source project. But the same can be archived by simply sending Pushbullet or Pushover notifications out, because those services are hooked into IFTTT. This would allow you to not only keep on supporting TP-Link devices, but also other platforms of smart plugs w/o interacting with their API directly.

jneilliii, 2021-12-12:
That issue was only a problem for a short period of time and only impacted UK devices as far as I know. If TPLink does ever fully block local communication then I will just stop working on the plugin and focus on my others that are local only (ie Tasmota). There are of course other options with MQTT and Home Assistant, etc. Thanks for the feedback though.

jneilliii, 2021-12-12:
BTW, there is an IFTTT plugin as well that you could utilize to control devices I suspect.

closed 2021-12-12 by jneilliii