jneilliii/OctoPrint-TPLinkSmartplug #292: Shut down octoprint

Biz810, 2021-12-30:
Hi there, 

Is it possible that you first shut down Octoprint with your plug-in and then switch off the socket with a time delay?

jneilliii, 2021-12-30:
That is achievable with the correct combination of settings. I documented an example in my Tasmota plugin, which is similar. Basically use countdown timers and system commands with the correct delays to achieve this. 

https://github.com/jneilliii/OctoPrint-Tasmota#examples

jneilliii, 2026-10-18:
As of version 2.0.0 of this plugin, countdown timers are no longer an option for this. Unfortunately, if the device shuts down there is nothing there to communicate the shut off command to the plug.

closed 2026-10-18 by jneilliii