jneilliii/OctoPrint-TPLinkSmartplug #293: [BUG]: Cannot connect to plu

mnunez021, 2022-01-12:
I just received my plug. it connects to the app, and google home without an issue, but not Octoprint. I can ping it from Octoprint and I added a route because it is on a different network but the same router. Here are my logs

[2022-01-12 19:51:10,759] DEBUG: IP 10.0.3.198 is valid.
[2022-01-12 19:51:10,760] DEBUG: Sending command {'system': {'set_relay_state': {'state': 0}}} to 10.0.3.198
[2022-01-12 19:51:13,852] DEBUG: Could not connect to 10.0.3.198.
[2022-01-12 19:51:13,853] DEBUG: None
[2022-01-12 19:51:41,518] DEBUG: resetting idle timer due to ClientOpened event.
[2022-01-12 19:51:41,930] DEBUG: abortTimeout: 30
[2022-01-12 19:51:41,949] DEBUG: powerOffWhenIdle: True
[2022-01-12 19:51:41,984] DEBUG: idleTimeout: 30
[2022-01-12 19:51:42,003] DEBUG: idleIgnoreCommands: M105
[2022-01-12 19:51:42,004] DEBUG: idleTimeoutWaitTemp: 50
[2022-01-12 19:51:42,012] DEBUG: powering on due to startup.
[2022-01-12 19:52:21,662] DEBUG: resetting idle timer due to ClientOpened event.
[2022-01-12 19:52:24,831] DEBUG: ImmutableMultiDict([('checkStatus', '10.0.3.198')])
[2022-01-12 19:52:24,832] DEBUG: Checking status of 10.0.3.198.
[2022-01-12 19:52:24,832] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-01-12 19:52:24,833] DEBUG: IP 10.0.3.198 is valid.
[2022-01-12 19:52:24,834] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.3.198
[2022-01-12 19:52:27,832] DEBUG: Could not connect to 10.0.3.198.
[2022-01-12 19:52:27,832] DEBUG: 
[2022-01-12 19:52:27,833] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-01-12 19:54:12,598] DEBUG: resetting idle timer due to ClientOpened event.
[2022-01-12 19:54:13,583] DEBUG: ImmutableMultiDict([('checkStatus', '10.0.3.198')])
[2022-01-12 19:54:13,583] DEBUG: Checking status of 10.0.3.198.
[2022-01-12 19:54:13,583] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-01-12 19:54:13,584] DEBUG: IP 10.0.3.198 is valid.
[2022-01-12 19:54:13,584] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.3.198
[2022-01-12 19:54:16,581] DEBUG: Could not connect to 10.0.3.198.
[2022-01-12 19:54:16,582] DEBUG: 
[2022-01-12 19:54:16,583] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-01-12 19:54:18,917] DEBUG: ImmutableMultiDict([('checkStatus', '10.0.3.198')])
[2022-01-12 19:54:18,918] DEBUG: Checking status of 10.0.3.198.
[2022-01-12 19:54:18,918] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-01-12 19:54:18,919] DEBUG: IP 10.0.3.198 is valid.
[2022-01-12 19:54:18,919] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.0.3.198
[2022-01-12 19:54:19,581] DEBUG: Could not connect to 10.0.3.198.
[2022-01-12 19:54:19,582] DEBUG: 
[2022-01-12 19:54:19,582] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}

jneilliii, 2022-01-12:
Since they are on separate networks I assume there is also security ACLs between those networks? If so, you may need to also allow the custom ports used by TPLink (9999). Otherwise, I'm not sure why it would not allow the local connection unless having it in cloud access mode breaks local connectivity. I believe I have read that before. What model plug and firmware version are you running?

mnunez021, 2022-01-12:
im running 1.3.3 I just purchased it today. I added a Firewall rule to see if it help but nope. I don't have any restrictions between the subnets so that shouldn't be the issue

jneilliii, 2022-01-12:
What model plug?

mnunez021, 2022-01-12:
EU TAPO P100

jneilliii, 2022-01-12:
Yeah, TAPO devices are not supported by my plugin. You might want to look at the fork https://github.com/Tyfy/OctoPrint-TapoSmartplug for your device.

closed 2022-01-12 by jneilliii

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.