jneilliii/OctoPrint-TPLinkSmartplug #295: @TPLINKOFF responding with none

Killerherts, 2022-01-28:
m80/m81 or @tplinkoff seem to run the command but nothing ever happens
```printer","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"C0:C9:E3:19:76:4D","updating":0,"led_off":0,"relay_state":1,"on_time":3932,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2022-01-28 18:58:08,279] DEBUG: TIM
[2022-01-28 18:58:08,341] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2022-01-28 18:58:39,188] DEBUG: Received TPLINKOFF command, attempting power off of 10.16.69.112.
[2022-01-28 18:58:39,196] DEBUG: None
[2022-01-28 19:00:13,516] DEBUG: Received TPLINKOFF command, attempting power off of .
[2022-01-28 19:00:13,523] DEBUG: None
[2022-01-28 19:03:36,335] DEBUG: resetting idle timer due to ClientOpened event.
[2022-01-28 19:03:36,826] DEBUG: abortTimeout: 30
[2022-01-28 19:03:36,854] DEBUG: powerOffWhenIdle: True
[2022-01-28 19:03:36,855] DEBUG: idleTimeout: 15
[2022-01-28 19:03:36,862] DEBUG: idleIgnoreCommands: M105
[2022-01-28 19:03:36,863] DEBUG: idleTimeoutWaitTemp: 40
[2022-01-28 19:04:14,684] DEBUG: resetting idle timer due to ClientOpened event.
[2022-01-28 19:04:16,017] DEBUG: ImmutableMultiDict([('checkStatus', '10.88.88.112')])
[2022-01-28 19:04:16,018] DEBUG: Checking status of 10.88.88.112.
[2022-01-28 19:04:16,018] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-01-28 19:04:16,019] DEBUG: IP 10.88.88.112 is valid.
[2022-01-28 19:04:16,019] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.88.88.112
[2022-01-28 19:04:16,034] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.0.4 Build 210414 Rel.190620","hw_ver":"3.0","model":"HS103(US)","deviceId":"80065EB387BDF96E8648AC1C337511651DDC1DCC","oemId":"1925575B7899062F242F15F7AFE4D486","hwId":"80CB7367CEA6E3B6DC96C2855EE7C79A","rssi":-40,"longitude_i":-1213273,"latitude_i":387053,"alias":"3d printer","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"C0:C9:E3:19:76:4D","updating":0,"led_off":0,"relay_state":1,"on_time":4300,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2022-01-28 19:04:16,036] DEBUG: TIM
[2022-01-28 19:04:52,580] DEBUG: Received M81 command, attempting power off of 10.16.69.112.
[2022-01-28 19:04:52,582] DEBUG: None
[2022-01-28 19:08:09,574] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2022-01-28 19:08:17,710] DEBUG: Received M81 command, attempting power off of 10.16.69.112.
[2022-01-28 19:08:17,713] DEBUG: None
[2022-01-28 19:08:23,305] DEBUG: Received M81 command, attempting power off of 10.16.69.112.
[2022-01-28 19:08:23,307] DEBUG: None
[2022-01-28 19:08:37,451] DEBUG: Received TPLINKOFF command, attempting power off of 10.16.69.112.
[2022-01-28 19:08:37,453] DEBUG: None
```
The debug error doesn't seem to give me a ton of information to go on.

jneilliii, 2022-01-28:
Dis you include the ip of the device with the command and enable gcode trigger for the device in settings?

Killerherts, 2022-01-28:
> Dis you include the ip of the device with the command and enable gcode trigger for the device in settings?

Yes I have. I also tried using @TPLINKOFF xx.xx.xx.xx

jneilliii, 2022-01-28:
So the one that seems to respond in that log above is `10.88.88.112`, whereas the commands appear to be trying to control `10.16.69.112`? Are you controlling multiple devices? Please share screenshots of your settings.

Killerherts, 2022-01-29:
Wow I'm pretty slow today sorry mate

closed 2022-01-29 by Killerherts