jneilliii/OctoPrint-TPLinkSmartplug #297: [BUG]: Spotty Connection issue  to smart outlet

thespica93, 2022-02-27:
**Describe the bug**
I have as Kasa TP-Link Outlet, It is assigned a static IP address in my router. I can ping the device. But when entering the IP into the plugin, I can't control the outlet

**To Reproduce**
Steps to reproduce the behavior:
1. Assign a static IP to the plug
2. Install plugin
3. Configure the plugin
4. See error

**Expected behavior**
Assigning the plugin the same IP address as the outlet should give me control in Octoprint

**Desktop (please complete the following information):**
 - OS: MasOS Montery 12.1
 - Browser Chrome
 - OctoPrint Version: Version 1.7.3
 - Plugin Version- not sure where to find this

jneilliii, 2022-02-27:
It's possible that your TPLink firmware is not allowing the local control. Enable debug logging in the plugin's settings, restart OctoPrint and let's see what the log says. It will be saved in the logging section of OctoPrint as plugin_tplinksmartplug_debug.og.

thespica93, 2022-02-27:
Thank you! I’ll try that now.

thespica93, 2022-02-27:
Here is the log. It started working where I could turn it on or off. But as
soon as a print would start warming up, it shut the printer off
Thank you
-Nick

jneilliii, 2022-02-27:
Attachments via email don't come through, you'll need to drag it into a comment on github. But your description of it shutting down sounds like maybe the thermal runaway settings may be too low or incorrect.

thespica93, 2022-02-28:
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/8149961/plugin_tplinksmartplug_debug.log)
[image]

Here is the log and a screenshot of the current settings. I deleted the outlet currently because that was easier than removing the outlet from the wall, thats why you don't see the outlet configured

Thank you!

jneilliii, 2022-02-28:
Yeah, I think you have the temps reversed. 75 degrees is super low for extruder temp...

thespica93, 2022-02-28:
I am so sorry! I can't believe I screwed that up. It was one of those days I guess. I can't believe I have those numbers reversed. I feel so dumb. 
I'll let you know if I still have connection issues. 
Thanks again

thespica93, 2022-03-04:
Issue is still persisting with spotty connection. I am uploading the logs after I have tried to turn the printer on
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/8188984/plugin_tplinksmartplug_debug.log)

jneilliii, 2022-03-04:
Log is littered with `Could not connect to 192.168.1.205.` are you sure the IP address didn't change on you? Only other time I've seen this kind of error was when the firmware prevented local external control (see pinned post) but that only effected UK models at that time.

thespica93, 2022-03-04:
Yep the plug is at 192.168.1.205. I was able to ping it. As well as I
checked my router and verified its IP address. I set it to a static address
in the router. Maybe it is the firmware

jneilliii, 2022-03-04:
Couple of things you could try if you haven't already, unplug the tplink from the wall and plug it back in and possibly restart your router.

closed 2022-03-26 by github-actions[bot]