jneilliii/OctoPrint-TPLinkSmartplug #299: [Feature Request]: P100 plug support.

dannyglover, 2022-03-21:
**Is your feature request related to a problem? Please describe.**

P100 are popular plugs and support for them would be great.

**Describe the solution you'd like**

I found a python implementation that supports Tapo P100 plugs.
Implement P100 plug support using the work started by K4CZP3R > https://github.com/K4CZP3R/tapo-p100-python

jneilliii, 2022-03-21:
Just use the fork, I'm not implementing this support.

closed 2022-03-21 by jneilliii

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.