jneilliii/OctoPrint-TPLinkSmartplug #301: [BUG]: Octoprint never connects to my printer when they are both powered by TPLink plugs

moccor, 2022-04-06:
**Describe the bug**
Octoprint never connects to my printer when they are both powered by TPLink plugs. I can unplug my printer and plug it directly into the outlet, and everything will work fine with Octopi connecting automatically. But when I plug the printer into a plug, for some reason it will never automatically connect and never allow me to connect if I try manually (it will give an error).

**Expected behavior**
Raspberry Pi should turn on, printer should turn on after the delay, Pi should connect to printer after a delay, and then worst case scenario manual connect should work, but it will never work. Works fine when the printer isn't in the smart plug.

**Desktop (please complete the following information):**
 - OS: W10
 - Browser: Chrome
 - OctoPrint Version: Latest
 - Plugin Version: Latest

**Additional context**
I can just plug the printer in without a TPLink smart plug and it will work fine. The plugs turn on fine. The only thing that doesn't work is the "connecting" to the printer part for some reason.
[image]
[image]
[image]

jneilliii, 2022-04-06:
You need to set the plug that the printer is connected to for auto-connect, not the pi. Also, in the connection panel of OctoPrint make sure you have specifically selected your port and baudrate and select the option to save the settings.

moccor, 2022-04-12:
Alright, thanks for the info. When I tried to use it yesterday, I kept getting a new problem. Whenever the plug turns on from TPLinkSmartplug Editor, I get this error : 

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3779

But the USB0 and 115200baudrate work fine. Today I disabled the printer from turning on automatically from the plugin and instead did it manually from the button at the top, and it worked fine with those two saved settings.

jneilliii, 2022-04-12:
Try increasing the auto-connect delay.

moccor, 2022-04-12:
I changed it from 30 to 60 and still gives the same error. Since I only did it once, I noticed there is a lot more info at the top of the terminal, maybe some of it is useful. It is weird cause at one point it says it is operational :

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6983faf0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=20.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3779
Recv: JumpToApp0
Recv: JumpToApp1
Recv: MACHINE_TYPE:A30T UUID:\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff FIRMWARE_NAME:V1.xx.03
Recv: PROTOCOL_VERSION:V1.0 EXTRUDER_COUNT:3
Recv: BOARD_INFOR:Smartto MB V1.2
Recv: echo:SD init fail
Recv: echo:Unknown command:
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Recv: ok
[...]
Send: N0 M110 N0*125
Recv: ok
Send: M115
Recv: ok
Send: M21
Recv: MACHINE_TYPE:A30T UUID:\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff FIRMWARE_NAME:V1.xx.03
Recv: PROTOCOL_VERSION:V1.0 EXTRUDER_COUNT:3
Recv: BOARD_INFOR:Smartto MB V1.2
Recv: ok
Recv: echo:SD init fail
Recv: ok
[...]
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3779

moccor, 2022-05-02:
So yesterday I changed it from 60 seconds to 30, saved, and just did some normal printing. Turned it off as usual. Today I turned it on, and weirdly, it auto connected fine. But it never wants to shutdown for me after the specified time. So I enabled logging, restarted Octoprint (ofc it shuts down the plug then lol). I turn it back on, auto connect doesn't work and connecting gives that error above. I don't see any pattern to this, but it is weird how it somehow worked once for the autoconnect. And I assume there is no way to get a good log of it, or is there?

jneilliii, 2022-05-02:
enabling debug logging in the plugin's settings and restarting OctoPrint will help in getting to the bottom of what the plugin is doing, or trying to do. It will generate a separate plugins_tplinksmartplug_debug.log file that you can share here and I can try to deduce what's happening. I suspect that options enabled on one device are causing the other device to get triggered or vice versa.

rashuge, 2023-03-27:
I am seeing an issue as well where my printer will not turn on when I send something from Cura connected to OctoPrint.  Similar to what is happening here.  I get the auto shutdown to happen just fine.  But the turn on of the printer does not work, the auto connect works if I manually power the printer but that is due to OctoPrint I assume.  I will try to get some debug logs happening and submit those.

jneilliii, 2023-03-27:
When your printer is powered off does the screen stay on our have you already mitigated back powering issues from the pi?

rashuge, 2023-03-27:
I have custom created a cable that mitigates the 5v back feed for my Creality.  It is ONLY powered by the TP-Link plug.

rashuge, 2023-03-27:
[image]
[image]

jneilliii, 2023-03-27:
With Cura, the plugin inside Cura controls powering on is what I was trying to say. Open Cura, and where you configured it's plugin for sending to OctoPrint there will be another option for automatically powering on using the selected plug. I'm not going to be at my computer for awhile to take a screenshot to show you, but can when I get home after work.

rashuge, 2023-03-27:
I totally missed that.  Sorry to have wasted your time.