jneilliii/OctoPrint-TPLinkSmartplug #305: [Feature Request]: Navbar dropdown instead of icons for each plug on multiple plug devices

Ragnar123456, 2022-04-17:
For multiple plug smartplugs, the UI currently takes up a lot of space on the navbar. I would like to see an an option to have the plugin have a single dropdown icon on the navbar with the specific plugs shown when the dropdown is clicked. The current Enclosure plugin is an example of how this might be done.

jneilliii, 2022-04-23:
Added in above commit and available in version 1.0.3rc1. If you want to try it out, change the release channel for the plugin in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

Ragnar123456, 2022-04-24:
Downloaded the RC. Dropdown works as expected. Nothing appears broken after a few minutes of testing.

closed 2022-04-24 by jneilliii