jneilliii/OctoPrint-TPLinkSmartplug #306: Graphs not appearing

acer8888, 2022-04-26:
The graphs showing the power usage dont appear for me. I have a tab available for the graphs, just get a small white space. Any suggestions to fix?

acer8888, 2022-04-27:
> Is status polling enabled?

I double checked, yes its set to every 1 minute, as well as during prints.

jneilliii, 2022-04-27:
What model device do you have? Can you enable debug logging, restart octoprint and gather some data for a while to see if it's getting the correct info?

acer8888, 2022-04-27:
I have 2 HS100 US, I'm in the middle of a long print atm, but i can definitely enable debug logging and get that data uploaded tomorrow. Thank you i appreciate the support!

jneilliii, 2022-04-27:
Ah, the HS100 I don't think includes power monitoring, so the tab shouldn't even be showing up for you.

acer8888, 2022-04-27:
You are correct, my apologies I was looking at the specs for the hs110 models. The hs100 that I have does not have power monitoring.

closed 2022-04-27 by acer8888