jneilliii/OctoPrint-TPLinkSmartplug #310: [Feature Request]: Schedule Power On/Autoconnect when PowerOn detected

Essy1245, 2022-05-22:
**Is your feature request related to a problem? Please describe.**
Not related to any Github issues. 
I sometimes use Print Scheduler (one of your plugins) but I want it fully automated. The printer needs to be turned on and connected before the scheduled print starts, making the schedule function not useful. 

**Describe the solution you'd like**
Either schedule power on (Assuming Auto-connect will work), or detect if the plug is "on" and connect (so even when I turn the plug on outside octoprint it still connects. Planning to schedule turn-on time via TPLink Kasa app). 

**Describe alternatives you've considered**
I tried scheduling power on via TPLink Kasa app, but I still have to hit connect manually since OctoPrint has no idea if it's on or off.
Tried ["Helpful System Commands"](https://github.com/jneilliii/OctoPrint-PrintScheduler/wiki/Helpful-System-Commands) in PrintScheduler's Wiki. The simple answer is it doesn't work. Will mention why in Additional context below.

**Additional context**
Ender 3 w/ Klipper @RPI3B+.
I'm running Klipper so I have to issue "RESTART" (as gcode) after good 15 seconds after the printer is powered on. 
Prints can't be started until I receive `Recv: // Klipper state: Ready` in serial. 
What happens when I turn on the printer with system commands mentioned above?
1. The printer turns on
2. Klipper starts getting ready
3. PrintScheduler plugin spams "GCODE Before Each Print" regardless of Klipper's state.
4. Print starts
5. Klipper cancells the print because it's (or the connection) is not ready yet.
6. I get `Recv: // Klipper state: Ready` in serial.
7. (And then I can start the print manually.)

I know this request might belong to PrintScheduler's repo but I thought this feature was specific to this plugin so I submitted it here.

jneilliii, 2022-05-22:
Have you tried adjusting any of the delay options? Is your restart command configured in the gcode after on option? Maybe share a screenshot of your settings would be helpful. It might be possible to put the restart in this plugin's settings

jneilliii, 2022-05-22:
BTW, it might be possible to use portlister plugin for auto connect on serial port detection. just not sure if it's compatible with klipper.

Essy1245, 2022-05-22:
> Have you tried adjusting any of the delay options? Is your restart command configured in the gcode after on option? Maybe share a screenshot of your settings would be helpful. It might be possible to put the restart in this plugin's settings

It waits for 10s before connecting (but I leave Klipper connected all the time, so this does nothing) and then sends restart command.
Need to wait for about 5 seconds to get `Recv: // Klipper state: Ready` and then I can start prints. 

TPLinkSmartplug config:
[image]

PrintScheduler config:
[image]

Octoprint config (Gcode before print. Gcode after "connection to printer is established" is blank)
[image]

> BTW, it might be possible to use portlister plugin for auto connect on serial port detection. just not sure if it's compatible with klipper.

I have tried portlister, but since "connect" in Octoprint connects to "Klipper" instead of the actual printer (and Klipper connects to printer's mcu after Klipper is connected to Octoprint), it did not work. 
I leave Klipper connected all the time. I made this plugin to send a restart command 10 seconds after the plug is turned on. 
Klipper attempts to talk with the printer after that. It takes about 5 seconds until I get `Recv: // Klipper state: Ready`.

Essy1245, 2022-05-22:
Also, few quick questions here.
1. Where can you find the graph like shown [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/master/screenshot_tab.png)? It is not in the plugins tab or settings tab.

2. What Gcode commands can I send to turn the plug on/off? (if I have `GCODE trigger` enabled) It is not specified anywhere.
I tried M80 and M81 but looks like it's specific to Marlin.

jneilliii, 2022-05-23:
> 1. Where can you find the graph like shown [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/master/screenshot_tab.png)? It is not in the plugins tab or settings tab.

It will automatically show if your Kasa device supports power monitoring. Not all devices do.

> 2. What Gcode commands can I send to turn the plug on/off? (if I have GCODE trigger enabled) It is not specified anywhere.
I tried M80 and M81 but looks like it's specific to Marlin.

See [settings explained](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug#settings-explained), in short, `M80/M81 <plug ip here>` or `@TPLINKON/@TPLINKOFF <your ip here>`.

jneilliii, 2022-05-23:
I wonder if you could add `&& sleep 45` to your system command in print scheduler to sleep 45 seconds after the curl command is sent. That might work around the issue here.

Essy1245, 2022-05-24:
I tried `&& sleep ?` command after few minutes of Googling. Wasn't sure about the results tho, but I don't think it worked well.
I tested it with 60s, but it didn't work somehow. I think it waited 60 seconds in the background (assuming it ignored that command or it waited in the background).

>See [settings explained](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug#settings-explained), in short, M80/M81 <plug ip here>

I forgot the iP address :/