jneilliii/OctoPrint-TPLinkSmartplug #311: [Feature Request]: Clear data from selected and/or all plugs

Gozania, 2022-05-23:
I have since re-ordered the devices plugged in to help relieve cable strain.  The plug my printer was in is no longer used, yet the is still data from when first deployed.

Would love the ability to clear data from an individual plug or all plugs.

[image]

Gozania, 2022-05-23:
Thank you so much for being receptive,  Really enjoy the plugin and have been using it for some time now.  Even bought a 2nd strip for the next printer I get in preparation.

Gozania, 2022-07-02:
Curious if there was any progress on this?

jneilliii, 2022-07-02:
I have not had a chance to mess with it, been busy with the day job and a couple of other paying plugins at the moment.