jneilliii/OctoPrint-TPLinkSmartplug #312: [Feature Request]: Allow devices to be addressed by name, not just IP address

puterboy, 2022-05-24:
I have given all my plugs "names" in my router - so that I can address them by name which is helpful if the IP address changes.
Any reason why this can't be enabled and allowed here?

jneilliii, 2022-05-25:
in theory if the name is resolvable by the pi the plugin is running on it should work. I have the setting to try and figure out the ip from hostname [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/d4c66429cecc683e09d5dee5fcac4004635462eb/octoprint_tplinksmartplug/__init__.py#L1046)

puterboy, 2022-05-25:
But the GUI seems to only accept numerical IP addresses -- anything else shows up as 'red' print...

jneilliii, 2022-05-25:
Does it prevent you from saving?

puterboy, 2022-05-25:
I hadn't tried because it was marked in 'red' and I thought it was an error.
However, it does work nonetheless when I save it.
Perhaps update the plugin to allow for valid hostnames?

jneilliii, 2022-05-25:
Yeah, I could remove the red highlighting from the field to counter that confusion and update the label. There is no good way to validate hostnames, so will depend on the status check to throw that error.

puterboy, 2022-05-26:
Typically, it's up to the user to use a valid hostname. You could do some simple syntactic checks to make sure the hostname is consistent with the relevant RFC (maybe RFC 1035) -- e.g, no '@'

b-morgan, 2022-07-16:
I would like to see this implemented as well. In addition, a notes field would be useful. I was going to ask about using the MAC address (which I'd put in the notes if it wasn't entered somewhere else. The reason I ask is because my cable modem/router died and I had to replace it. All of the DHCP obtained IP addresses changed because the new modem/router uses a different net block (used to be 192.168.0.0/24, is now 10.0.0.0/24).

jneilliii, 2022-07-17:
@b-morgan I see you found the other FR about MAC Address addressing, but you can use hostname in the ip field even though it highlights it as red, di you happen to try that? I haven't heard back from anyone else if it works as expected.

puterboy, 2022-07-17:
Yes it works (despite the red). Sorry for not confirming earlier with you.
jneilliii wrote at about 00:09:37 -0700 on Sunday, July 17, 2022:
 > @b-morgan I see you found the other FR about MAC Address addressing, but you can use hostname in the ip field even though it highlights it as red, di you happen to try that? I haven't heard back from anyone else if it works as expected. 
 > 
 > -- 
 > https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/312#issuecomment-1186425560
 >

b-morgan, 2022-07-17:
Unless I can figure out how to name the plugs, each plug's name is the model name of the plug, i.e. HS105, HS103, and HS300 so while I tried the name, it didn't work.

b-morgan, 2022-07-17:
I did some research and changing the hostname of a plug doesn't appear to be possible. Issue #76 appears to be the best option. Alternatively, adding a "note" field to each entry would allow documenting the MAC address which could then be used to obtain the IP address manually.

jneilliii, 2022-07-17:
Thanks @b-morgan, obviously for those that only have one plug they can use hostname. Unfortunately, I never really found a good MAC to IP address translation option for python that wasn't a full network scanning/hacking tool which is why the option hasn't been implemented.

b-morgan, 2022-07-17:
How about adding a text "note" field to each entry. You don't have to do anything with it.

jneilliii, 2022-07-17:
Yes, I can definitely do that which is one reason I was leaving this issue open (that and red highlighting/label for IP). I was just looking at the original source of the encryption/decryption I used for this from softScheck and there is a [PR](https://github.com/softScheck/tplink-smartplug/pull/45) that implements a broadcast option. I could probably implement something like that to create a discovery of devices and create a lookup table based on the "info" command to all plugs in the network. Would probably take me a while to program that and test though. I think the wemo library I use for that plugin incorporates a similar approach for device discovery.

rushl, 2023-11-09:
What about referring to a plug by an ID set by the user? That avoids all lookup issues. Referring to plugs is SUPER useful, but going by IP is fragile. If I have to change an IP, I have to remember everywhere I referred to that plug to update it. But if I could give the plug an ID like "enclosure_lights" and just refer to that, I can update the plug's IP in the plug config, and everywhere the plug is referenced will continue to function.