jneilliii/OctoPrint-TPLinkSmartplug #313: [BUG]: Version 1.02 broke connectivity with HS103

TerraVestra, 2022-08-06:
The update to 1.0.2 on both of my printers broke connectivity to the TP-Link HS103.
I have verified that the HS103 is working through the Kasa App and its IP Address is pingable.

I've used your plugin for years and it's been amazing. Please assist with resolving this update-related issue.

Logs are:
[2022-08-06 10:54:39,832] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-08-06 11:09:39,839] DEBUG: Checking status of 192.168.1.206.
[2022-08-06 11:09:39,840] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-08-06 11:09:39,841] DEBUG: IP 192.168.1.206 is valid.
[2022-08-06 11:09:39,841] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.206
[2022-08-06 11:09:42,946] DEBUG: Could not connect to 192.168.1.206.

jneilliii, 2022-08-06:
Nothing on the communication/control side was changed between 1.0.1 and 1.0.2. Based on the above it looks like your plug isn't responding to the "get status" command. When this has happened before it was because of UK models getting firmware updates that blocked local direct communication. What version firmware is running on your HS103? Can you also verify from your router that the IP address for the device didn't change.

TerraVestra, 2022-08-06:
I have verified that the IP address of the plug hasn't changed. It's set to a static IP address and I can ping it when it's on and control it with the kasa app.

So I uninstalled 1.0.2 and removed the plugin data, then reinstalled 1.0.1. Turns out - that still doesn't work. It literally just worked before the update. When I did the update though, updated your plugin along with octoprint software (I was behind a few versions). It's probably safe to conclude that this isn't a result of anything in 1.0.2 but would you have any other ideas on what could be causing this problem?

TerraVestra, 2022-08-06:
BTW:
Model: HS103
HW Version: 5.0
FW Version: 1.0.3

TerraVestra, 2022-08-06:
You know what, it turned out to be a networking issue afterall. I've reinstalled 1.0.2 and it works great.
Looks like it was a stupid oversight on my end. Router IP Assignments got cleared and some other device tool over its IP address. The timing of the initial update was purely coincidental.

Thank you for your help!

jneilliii, 2022-08-06:
Yeah, that's what I was thinking might have happened. Thanks for reporting back.

closed 2022-08-06 by jneilliii