jneilliii/OctoPrint-TPLinkSmartplug #314: [Feature Request]: Ability to turn off Plug based on other sensor data (e.g., enclosure temperature

puterboy, 2022-08-07:
Hi @jneilliii -- really love this plugin...
And now that I have gotten my enclosure sensor to display in your other Plotly plugin, I would like to use the enclosure sensor to trigger a TP-Link power off if the temperature exceeds a user-specified value, similar to what you can do with the existing sensors.

I see this as a potentially important safety measure in that if the chamber is overheating, I can kill power to the printer and accessories, even if the other sensors have not yet triggered.

How hard would this be to implement?

jneilliii, 2022-08-08:
It would probably make more sense to keep the logic of the sensor reading and off triggers for those as part of the single file plugin that you made for the PlotyTempGraph plugin. In this plugin's current state there are only a couple of ways to trigger the off command. That is via API call as documented in the wiki or one of the gcode trigger commands when that option is enabled. I have been considering to add both CLI and [plugin helpers](https://docs.octoprint.org/en/master/plugins/helpers.html) to make integrating a little easier for others.

puterboy, 2022-12-28:
I assume you mean adding logic to the PlotlyTempGraph plugins that trigger the TP-Switch to turn on/off. Question is how do I call the TP Link on/off switch from PlotlyTempGraph. I could hack something based on some other python code I have for accessing TP Link switches but that seems messy. But if I have to do so much hacking, why even use Octoprint, I could just write a shell script to test a sensor value and based on that trigger turn on/off TP Link plugs using totally external code. 

A more important problem with PlotlyTempGraph is that it only monitors temps when the printer is on while one might want to cut power to the plug (or alternatively turn it on) based on a sensor value even when the Printer is off.

Wouldn't it be easier to just have a text box added to this plugin that is sent as a system (shell) command returning 0/1 -- where 1 means turn off (0 means do nothing). The command could be just about anything, allowing for triggers based on any external sensor event or even other conditions (e.g., time/date, status of another program, etc.). Seems dead simple, to exec the shell command every X seconds (configurable?) and then trigger switch based on return value.
(Alternatively the system command could return a temperature and then based on that temperature you could set a trigger to turn off the plug -- but this seems less general since not all events may be temperatures).

If you want to be fancy, you could:
1. Have an option to reverse the logic so that 1 turns on the plug (still 0 does nothing)  - or even logic where 1 turns on and 0 turns off
2. Allow multiple sensor toggle command text-boxes

jneilliii, 2026-10-18:
I've been thinking about this some more and believe the reading and reacting to sensor data is out of scope for this plugin. I know @jacopotediosi is in the process of reviving the enclosure plugin and wondering if I might be able to collaborate with him and provide the necessary helper functions in this plugin to allow that plugin to send on/off commands based on conditions. Would be happy to integrate that.

jacopotediosi, 2026-10-18:
Yes @jneilliii, if you add helper functions I can add a new output type in the enclosure plugin pointing to them.

jneilliii, 2026-10-22:
Version 2.0.1 was just released that expose turn_on, turn_off, and check_status via plugin helpers. They are direct pass-through commands to the equivalent [turn_on](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/85bd721222b533f9c7fa136071024651054680c1/octoprint_tplinksmartplug/__init__.py#L471), [turn_off](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/85bd721222b533f9c7fa136071024651054680c1/octoprint_tplinksmartplug/__init__.py#L504), and [check_status](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/85bd721222b533f9c7fa136071024651054680c1/octoprint_tplinksmartplug/__init__.py#L535) commands in the plugin. 

If there are any needed changes on my end to help with that integration, just let me know.

closed 2026-10-22 by jneilliii