jneilliii/OctoPrint-TPLinkSmartplug #315: [Feature Request]: Ability to "lock" printer on (or off), overriding conditions and timeouts

puterboy, 2022-08-07:
I typically use a 30-60min idle timeout but sometimes I would like to keep the printer on longer even when idle -- especially when I am testing something.... and it can be easy to miss the warning popups on the screen.

I could manually change the settings each time, but it would be nice if there were an override to keep the printer on (or off) regardless of other conditions (except perhaps thermal runaway). 

Perhaps a long press on the button could be used to trigger an override with the icon changing to show a green (or red) lock instead of the thunderbolt to indicate the locked state.

jneilliii, 2022-08-08:
There's a toggle switch in the sidebar to disable idle timeout.

puterboy, 2022-08-08:
Hmmm... Which sidebar?
I just see an on/off button in my navbar and then the detailed configs under settings.

jneilliii, 2022-08-08:
It looks different for me because I have UI Customizer and have it moved to the top, but it looks like this...

[image]

puterboy, 2022-08-08:
I only see the thunderbolt icon in my nav bar
[image]

I don't get anything in any of the tabs.

puterboy, 2022-08-08:
Never mind - I see it below the fold on the right nav. I just had never scrolled that low :)

closed 2022-08-08 by puterboy