jneilliii/OctoPrint-TPLinkSmartplug #316: [Feature Request]: Option to use the TPLink Cloud API instead of IP address

derekantrican, 2022-08-08:
**Is your feature request related to a problem? Please describe.**

For some reason, my TP Link doesn't appear connected to my router (can't see it using Netgear's built-in "attached devices" feature or using a network scanning app like Fling) but is still togglable using the Kasa app. This means that the plugin fails to access it via IP address. I've also used [this 3rd party Windows app](https://www.microsoft.com/store/productId/9PF9TCZDRKZD) successfully to toggle the state of the plug. I believe the app is using the API according to this post (which, of course, requires you to provide your authentication details according to tplinkcloud.com): https://itnerd.space/2017/06/19/how-to-authenticate-to-tp-link-cloud-api/

**Describe the solution you'd like**
I believe it would solve my problem if the app could use the cloud API & auth rather than just IP address, so I would like to see that as an alternative option.

jneilliii, 2022-08-08:
I won't do that because I'll never use the cloud service personally. I believe the other TP-Link plugin does already use the cloud API though, maybe it would work for you.

closed 2022-08-08 by jneilliii

derekantrican, 2022-08-18:
@jneilliii I'm working on a plugin for this (when I have time) and have gotten some basic functionality working. If I (eventually) made a PR in this repo to give users the optional ability to trigger switches via the cloud API rather than IP address, would you merge it in? I think users may be interested in all the other features that your plugin here provides (especially connection with Cura, which I'm not sure how to do) and I think it'd be better than having two almost-identical plugins that only use _slightly_ different methods

jneilliii, 2022-08-18:
I would most certainly accept a PR with this integration as an optional component.

jneilliii, 2022-08-18:
another option would be to create a PSU Control Sub-Plugin that is using the cloud API and honestly that might be easier than cludging through all my code and is supported by the Cura integration.

derekantrican, 2022-08-27:
@jneilliii I spent some time learning OctoPrint plugins and make a proof-of-concept: https://github.com/derekantrican/OctoPrint-KasaCloud . I also just threw together a quick mockup and wanted to get your thoughts (and feedback is welcome and appreciated):

[image]

[image]

jneilliii, 2022-08-27:
Your POC plugin doesn't look bad. There are a couple of points. Any requests.get or .post should have a timeout parameter set. I would probably [restrict the password field on the settings API](https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.SettingsPlugin.get_settings_restricted_paths). It would also be possible to hash out the password stored in settings. and example of that can be seen [here](https://github.com/eshapiro42/OctoPrint-Wyze/blob/3d12e13576fc9f2958686d726cffdfee0353824f/octoprint_wyze/__init__.py#L48-L79). I've been meaning to implement that in this plugin, just haven't gotten around to it yet. I would probably make the "list devices" part of the device editor window as a pull down to the device id field.