jneilliii/OctoPrint-TPLinkSmartplug #321: [BUG]: It doesn't turn off my printer anymore? HS100

lloyb, 2022-09-20:
I used to use this plugin to turn off my HS100 plug, I got a new Pi 4 so made a new octoprint install 
and when trying to set things up like before I installed this but the printer doesn't turn off, it just sits there.

Pressing the green lightning bolt turns the printer off so the plugin works
but any idea why it never sends an off command at the end of printing?

jneilliii, 2022-09-20:
Did you configure it to go off with idle timer options and make sure that option is checked in both the general settings and the individual plug's setting?

lloyb, 2022-09-26:
[image]
[image]

I went back in and reinstalled it, I ticked Off on Idle like you suggested and my printer still just sits on after a print. what am I doing wrong?

lloyb, 2022-09-26:
Just a follow up to say I've done two prints today, after the first I posted the images above then I've just done another and it turned the printer off. How bizare

jneilliii, 2022-09-26:
weird, if you made the change while the printer was still printing then it wouldn't have started the idle timer I suspect. even then, it's a 30 minute timeout, so you would have to wait at least that long after print completion before the idle would trigger.

lloyb, 2022-10-04:
Yeah it seems that it waits 30mins before turning off, I tried to change it to 1min but it still seems to wait 30mins, am I going crazy or shouldn't this plugin have a tickbox that says "Turn off printer after print job has finished"?

Thats all I want

jneilliii, 2022-10-04:
If you want to do that the best option is to enable the gcode trigger options and then add the appropriate command to your end gcode scripts in slicer or OctoPrint's gcode scripts section.

jneilliii, 2022-10-19:
Whacker debug logging in three plugins settings, restart OctoPrint and let it go through it's steps. Then upload the plugin_tplinksmartpkug_debug.log file from OctoPrint's logging section in settings.

closed 2022-11-11 by github-actions[bot]