jneilliii/OctoPrint-TPLinkSmartplug #322: Upload Event Monitor not working for me

Pete2345, 2022-10-02:
I am trying to have my printer turn up automatically when I upload a file but not print it.

I know "Upload Event Monitor" is designed to do this and I have enabled it in the global settings and also on for the specific plug but when I hit "Print with Octoprint" from Cura, It uploads the file to Octoprint but my TP link plug doesn't turn on.

If however I select "Always on upload" then the plug turns on but it starts to print automatically which I know ifs how it's supposed to work but I don't want this.

What am I doing wrong please?

[image]

jneilliii, 2022-10-02:
If you're using the OctoPrint Connection plugin in Cura then it can actually control powering on the plug if it detects this plugin installed. Unfortunately, seems it uses the same logic and that option is only enabled if you enable the option to start print immediately on upload. Without always on upload it will only power on if the option to start print is enabled in the upload process (Prusa/Super Slicer options). So as you've deduced already, the plugin is working as designed from a workflow perspective. I'm more interested in your use case as to why powering on with upload without starting the print would be something you would want to do?

Pete2345, 2022-10-02:
Thank you for the reply and it saves me spending too long problem solving.

My reasons are that I can spend a while finding or designing something then slicing it but when I want to print it I often need to change the fillament as I have the wrong one in more often than not.

I can of course power on the printer first, pre heat, change the fillament and then hit print but I thought it would be quite nice to

Hit print in Cura
Have the plug turn the printer on
Load the gcode ready
Preheat
Wait
I could then change the fillament and hit print.

I suppose what I could do is enable Startup event monitoring and add
M77: Stop Print
M104 S200: Heat hotend to 200.

This would then turn the printer on,  start the print but then stop it, preheat and then wait.

I'm printing at the moment so can't test it but in theory it could work. Or I could just add a pause into Cura start Gcode to pause after the pre heat

jneilliii, 2022-10-02:
Yeah, adding `@PAUSE` to your start gcode after heating would be an option. Would require you to resume from OctoPrint side if you do that method.

closed 2022-10-24 by github-actions[bot]