jneilliii/OctoPrint-TPLinkSmartplug #323: [BUG]:Using Octodash, I can power on a TP-Link Plug but not power it off

redjacketpress, 2022-11-01:
**Describe the bug**
While the TP-Link controls continue to work as expected in Octoprint, when accessing them through _Octodash,_ I can power on the plugs, though I cannot reliably power them off. When I say "reliably," I mean that power off used to happen within a few seconds -- it now seems to take more than a minute, and I'm not sure if the command is being processed, or if they're responding to the idle timeout (though I have my idle timeout set for 5 minutes, so that seems unlikely).

(These are Kasa HS105 devices, I have two of them.)

jneilliii, 2022-11-01:
Enable debug logging in the plugin's settings, restart octoprint and test some more. That will give us something to evaluate in plugin_tplinksmartplug_debug.log.

redjacketpress, 2022-11-01:
Sending the log file, as well as screenshots of my settings.

This time, it turned off the Printer's plug within about 30 seconds, though the Light's plug took several minutes. (My results have been inconsistent -- sometimes, the Light's plug will power off first, or _both_ plugs will take minutes to power off.)

Appreciate the help!

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/9914325/plugin_tplinksmartplug_debug.log)
[image]
[image]
[image]

jneilliii, 2022-11-02:
Are you using `[!POWEROFF]` for the custom command associated with the button or are you using gcode commands? The gcode ones won't work without a connection to printer, but with `[!POWEROFF]' that will control the one that is configured in OctoDash's settings using direct API calls. Neither of your plugs should be powering off on idle because you haven't enabled those options in the individual plug's settings.

redjacketpress, 2022-11-02:
I've been using a GCode command for the printer, M81 192.168.4.81; with @TPLINKON (or @TPLINKOFF) 192.168.4.82 for the light. Both are also used in Octoprint for "After print job completes."

(My Understanding was that [!POWEROFF] requires an entirely different plugin and hardware.)

When I activate Octodash it powers on the printer and establishes a connection. When I send the M81 192.168.4.81 command through Octodash, I can see that status -- powering off -- on my printer, though the plug doesn't power off quickly as it used to (and the printer just restarts, instead).

(I hadn't read the tooltip carefully enough to know I had to enable the "Idle" setting in the plugs as well. Thanks!)

redjacketpress, 2022-11-03:
After playing with this a bit more -- adjusting my settings for "GCODE Off Delay" for both Printer and Light to 5 seconds, and now sending a command for _both_ devices to power off at the same time -- the Light (TPLINK OFF) will power off in about 15-20 seconds, though the Printer (M1) requires about 60-70 seconds.

I can adjust the "GCODE Off Delay" for the Light plug to bring it closer when the Printer plug powers down, but the Printer plug still requires about a minute to power off.

Having had a look at the device information for both, I see the Printer plug is Hardware Verson 3.3 / Firmware 1.0.22, while the Light plug is Hardware Verson 4.0 / Firmware 1.0.3 (and both have current firmware). I think that might explain the discrepency in response time. So both seem to be working, just not as quickly as I had become accustomed to.

jneilliii, 2022-11-03:
Not sure why there would be a discrepancy between the hardware/firmware version differences. I personally use a power strip currently and it's been a while since I've tested newer devices. Are both devices local only or are they cloud connected?

redjacketpress, 2022-11-03:
Both local — I don’t think either have the ability to connect to a cloud service.

closed 2022-11-17 by redjacketpress