jneilliii/OctoPrint-TPLinkSmartplug #324: All switches stopped working last night

kevman28, 2022-12-08:
I have 4 printers with four Kasa devices ( DCHP static assignment) and none of them work today, here is an example log output , I have verified the IP of the Kasa device and this is occurring on 4 instances. Did something change on the TPlink side ??

022-12-08 01:16:11,667] INFO: Turning off 192.168.1.220 at 2022-12-08 01:16:11.667036
[2022-12-08 07:46:51,626] INFO: Turning off 192.168.1.220 at 2022-12-08 07:46:51.626499
[2022-12-08 08:40:17,885] INFO: Turning off 192.168.1.220 at 2022-12-08 08:40:17.885450
[2022-12-08 09:22:52,346] INFO: Turning off 192.168.1.220 at 2022-12-08 09:22:52.346403
[2022-12-08 10:39:03,985] INFO: Turning off 192.168.1.220 at 2022-12-08 10:39:03.985014
[2022-12-08 13:04:53,988] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.1.220')])
[2022-12-08 13:04:53,988] DEBUG: Checking status of 192.168.1.220.
[2022-12-08 13:04:53,988] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 13:04:53,989] DEBUG: IP 192.168.1.220 is valid.
[2022-12-08 13:04:53,989] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.220
[2022-12-08 13:04:57,084] DEBUG: Could not connect to 192.168.1.220.
[2022-12-08 13:04:57,085] DEBUG: 
[2022-12-08 13:04:57,085] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-12-08 13:04:57,116] DEBUG: Checking status of 192.168.1.220.
[2022-12-08 13:04:57,117] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 13:04:57,117] DEBUG: IP 192.168.1.220 is valid.
[2022-12-08 13:04:57,117] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.220
[2022-12-08 13:05:00,204] DEBUG: Could not connect to 192.168.1.220.
[2022-12-08 13:05:00,205] DEBUG: 
[2022-12-08 13:05:00,207] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}

jneilliii, 2022-12-08:
It is quite possible if your firmware auto-updated that TPLink is blocking local control now. That has happened in the past with some units in the UK.

jneilliii, 2022-12-08:
Based on the log output, basically the plugin sees the device connected, tries to get a current status update from the device and the device never responds, which is giving that `Could not connect to` message.

kevman28, 2022-12-08:
correct I have validated all the IPs are correct and it still works from the Kasa AP. Could it be a firmware change on the devices ? 

> Based on the log output, basically the plugin sees the device connected, tries to get a current status update from the device and the device never responds, which is giving that `Could not connect to` message.

jneilliii, 2022-12-08:
It could be possible. What hardware model device and what firmware version? [This ](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/218 )is an old thread in regard to when it happened before, it didn't only effect this plugin but larger projects like Home Assistant, https://github.com/home-assistant/core/issues/43088.

kevman28, 2022-12-08:
These are the devices and firmware that are impacted

hs103 V2.1 FIRMWARE 1.1.4
hs103 V3   FIRMWARE 1.0.4
HS103 V3   FIRMWARE 1.0.4
HS103 V2.1 FIRMWARE 1.1.4

For testing I added another swithc which was attached to a kasa power strip, that didn't work either.

> It could be possible. What hardware model device and what firmware version? [This ](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/218)is an old thread in regard to when it happened before, it didn't only effect this plugin but larger projects like Home Assistant, [home-assistant/core#43088](https://github.com/home-assistant/core/issues/43088).

jneilliii, 2022-12-08:
My power strip is still working for sure, running firmware version 1.0.19. I just tested adding a HS103 hardware version 3.0 firmware version 1.0.3 and it's still working as well.

jneilliii, 2022-12-08:
Ah, slightly different. My strip is HS300 hardware V1.0 firmware version 1.0.19.

kevman28, 2022-12-08:
so I added two more devices to the instance, one is the strip and a new one .229, the .229 device works, and is the save version and firmware  v3, 1.04.

I am puzzled...

`[2022-12-08 21:28:38,278] DEBUG: Could not connect to 192.168.1.228.
[2022-12-08 21:28:38,278] DEBUG:
[2022-12-08 21:28:38,279] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-12-08 21:28:38,307] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.1.196/2')])
[2022-12-08 21:28:38,307] DEBUG: Checking status of 192.168.1.196/2.
[2022-12-08 21:28:38,308] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:38,308] DEBUG: IP 192.168.1.196 is valid.
[2022-12-08 21:28:38,324] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:38,325] DEBUG: IP 192.168.1.196 is valid.
[2022-12-08 21:28:38,325] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.196
[2022-12-08 21:28:41,478] DEBUG: Could not connect to 192.168.1.196.
[2022-12-08 21:28:41,479] DEBUG: get_device_id response: False
[2022-12-08 21:28:41,479] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': [False]}} to 192.168.1.196
[2022-12-08 21:28:44,598] DEBUG: Could not connect to 192.168.1.196.
[2022-12-08 21:28:47,366] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.1.228')])
[2022-12-08 21:28:47,367] DEBUG: Checking status of 192.168.1.228.
[2022-12-08 21:28:47,370] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:47,370] DEBUG: IP 192.168.1.228 is valid.
[2022-12-08 21:28:47,370] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.228
[2022-12-08 21:28:50,518] DEBUG: Could not connect to 192.168.1.228.
[2022-12-08 21:28:50,518] DEBUG:
[2022-12-08 21:28:50,519] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-12-08 21:28:50,539] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.1.196/2')])
[2022-12-08 21:28:50,539] DEBUG: Checking status of 192.168.1.196/2.
[2022-12-08 21:28:50,540] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:50,540] DEBUG: IP 192.168.1.196 is valid.
[2022-12-08 21:28:50,556] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:50,557] DEBUG: IP 192.168.1.196 is valid.
[2022-12-08 21:28:50,557] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.196
[2022-12-08 21:28:53,638] DEBUG: Could not connect to 192.168.1.196.
[2022-12-08 21:28:53,638] DEBUG: get_device_id response: False
[2022-12-08 21:28:53,639] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': [False]}} to 192.168.1.196
[2022-12-08 21:28:56,758] DEBUG: Could not connect to 192.168.1.196.
[2022-12-08 21:28:57,528] DEBUG: Checking status of 192.168.1.228.
[2022-12-08 21:28:57,528] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:28:57,528] DEBUG: IP 192.168.1.228 is valid.
[2022-12-08 21:28:57,528] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.228
[2022-12-08 21:29:00,678] DEBUG: Could not connect to 192.168.1.228.
[2022-12-08 21:29:00,679] DEBUG:
[2022-12-08 21:29:00,679] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2022-12-08 21:29:00,709] DEBUG: Turning on 192.168.1.229.
[2022-12-08 21:29:00,710] DEBUG: {'autoConnect': True, 'autoConnectDelay': 10, 'autoDisconnect': True, 'autoDisconnectDelay': 0, 'automaticShutdownEnabled': True, 'btnColor': '#808080', 'countdownOffDelay': 1, 'countdownOnDelay': 1, 'currentState': 'unknown', 'displayWarning': True, 'emeter': {'get_realtime': {}}, 'event_on_disconnect': False, 'event_on_error': False, 'event_on_shutdown': False, 'event_on_startup': False, 'event_on_upload': True, 'gcodeCmdOff': False, 'gcodeCmdOn': False, 'gcodeEnabled': False, 'gcodeOffDelay': 0, 'gcodeOnDelay': 0, 'gcodeRunCmdOff': '', 'gcodeRunCmdOn': '', 'icon': 'fab fa-accusoft', 'ip': '192.168.1.229', 'label': 'printer', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': False, 'useCountdownRules': False, 'warnPrinting': False}
[2022-12-08 21:29:00,711] DEBUG: IP 192.168.1.229 is valid.
[2022-12-08 21:29:00,711] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}} to 192.168.1.229
[2022-12-08 21:29:00,991] DEBUG: «-ý"system":{"set_relay_state":{"err_code":0}}}
[2022-12-08 21:29:00,991] DEBUG: 0
[2022-12-08 21:29:00,992] DEBUG: Resetting idle timer since plug 192.168.1.229 was just turned on.
[2022-12-08 21:29:00,993] DEBUG: Checking status of 192.168.1.229.
[2022-12-08 21:29:00,993] DEBUG: {'system': {'get_sysinfo': {}}}
[2022-12-08 21:29:00,994] DEBUG: IP 192.168.1.229 is valid.
[2022-12-08 21:29:00,994] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.229
[2022-12-08 21:29:01,177] DEBUG: «@�"system":{"get_sysinfo":{"sw_ver":"1.0.4 Build 210414 Rel.190620","hw_ver":"3.0","model":"HS103(US)","deviceId":"80068611253A35E7CA205C57DB16C1491D8B61DC","oemId":"1925575B7899062F242F15F7AFE4D486","hwId":"80CB7367CEA6E3B6DC96C2855EE7C79A","rssi":-49,"longitude_i":-746255,"latitude_i":404646,"alias":"MINI","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"60:32:B1:E2:65:F6","updating":0,"led_off":0,"relay_state":1,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2022-12-08 21:29:01,179] DEBUG: TIM
[2022-12-08 21:29:06,771] DEBUG: Turning off 192.168.1.229.
[... cut ...]
[2022-12-08 21:29:07,318] DEBUG: «@�"system":{"get_sysinfo":{"sw_ver":"1.0.4 Build 210414 Rel.190620","hw_ver":"3.0","model":"HS103(US)","deviceId":"80068611253A35E7CA205C57DB16C1491D8B61DC","oemId":"1925575B7899062F242F15F7AFE4D486","hwId":"80CB7367CEA6E3B6DC96C2855EE7C79A","rssi":-51,"longitude_i":-746255,"latitude_i":404646,"alias":"MINI","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"60:32:B1:E2:65:F6","updating":0,"led_off":0,"relay_state":0,"on_time":0,"active_mode":"none","icon_hash":"","dev_name":"Smart Wi-Fi Plug Mini","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2022-12-08 21:29:07,320] DEBUG: TIM
`

kevman28, 2022-12-08:
Are there commands I can run to query at the command line in Python ? can you provide an example so I can troubleshoot somre more? Looks like 3 out of 6 switches work and some working/non are on the same firmware and same version.

jneilliii, 2022-12-08:
The original communication was pulled from [here](https://github.com/softScheck/tplink-smartplug). You would probably have to utilize one of the forks though for strip control, this one [here ](https://github.com/AMDHome/tplink-smartplug)seems to include that. Looks like there might be a possible new handshake protocol too, not sure if that might be related to your issue. That fork is found [here](https://github.com/watfordjc/tplink-smartplug/tree/issue-81/watfordjc-patch1).

kevman28, 2022-12-08:
Thank you so much. The mystery deepens, from the pi, I run the exact command that you used to query to a switch that returns an error and it work fine!! this is just annoying at this point.... 

` ./tplink_smartplug.py -t 192.168.1.228 -c info
Sent:      {"system":{"get_sysinfo":{}}}
Received:  {"system":{"get_sysinfo":{"sw_ver":"1.1.4 Build 210409 Rel.113427","hw_ver":"2.1","model":"HS103(US)","deviceId":"*******************************","oemId":"*******************","hwId":"*************","rssi":-43,**************************************}}}
`

> The original communication was pulled from [here](https://github.com/softScheck/tplink-smartplug). You would probably have to utilize one of the forks though for strip control, this one [here [...]

jneilliii, 2022-12-08:
do you have any errors in octoprint.log?

kevman28, 2022-12-08:
> do you have any errors in octoprint.log? looks like I do...see below

`2022-12-08 21:12:06,198 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:14:08,358 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:15:02,358 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:15:38,518 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:27:08,839 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:28:10,599 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:28:33,079 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:28:44,599 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 21:28:56,759 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status
2022-12-08 22:18:30,759 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin tplinksmartplug
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 570, in on_api_get
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 487, in check_status`

jneilliii, 2022-12-09:
might be easiest if you could share the whole file. you can drag the file into the comment window to attach.

MatchstickWorks, 2022-12-15:
Checking in on this, as well. I have the switch icons in my top bar and they've just been spinning for a while now. I can turn things off, but not on. This is the HS300 model and I've attached octolog, and the plugin's debugging log. Let me know if you need anything else
[plugin_tplinksmartplug_debug (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/10239904/plugin_tplinksmartplug_debug.1.log)
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/10239905/octoprint.1.log)

MatchstickWorks, 2022-12-15:
Hardware: 2.0
Firmware: 1.0.11

jneilliii, 2022-12-15:
Just checked your octoprint.log and seems there is an error related to the sqllite database used for power monitoring. 

```
2022-12-15 10:58:18,490 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin tplinksmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 158, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 581, in on_api_command
    response = self.turn_off("{ip}".format(**data))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 470, in turn_off
    return self.check_status(plugip)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 537, in check_status
    [plugip, today.isoformat(' '), c, p, t, v])
sqlite3.DatabaseError: database disk image is malformed
```

@kevman28 do you see a similar error in your octoprint.log?

jneilliii, 2022-12-15:
There are two ways we can try to resolve this error @MatchstickWorks. Either manually deleting the database over SSH and restarting OctoPrint, or clearing the settings from Plugin Manager completely using the eraser icon on the line item for this plugin, restarting OctoPrint and then re-entering all your plug info again. Since the SSH one doesn't lose settings these are the commands (assuming octopi image, adjust paths accordingly for manual install).

```
rm /home/pi/.octoprint/data/tplinksmartplug/energy_data.db
sudo service octoprint restart
```

MatchstickWorks, 2022-12-15:
I'll try that! Currently in a print, so I probably shouldn't restart the OP service just yet :)

kevman28, 2022-12-16:
> Just checked your octoprint.log and seems there is an error related to the sqllite database used for power monitoring.
>
> ```
> 2022-12-15 10:58:18,490 - octoprint.server.api - ERROR - Error while [...]

yes- it was in reference to the same script __init__.py

kevman28, 2022-12-16:
i did some testing with the original communication script, looks like port 9999 no longer works....

`pi@octopi3:~/tplink-smartplug $ ./tplink_smartplug.py -t 192.168.1.238 -c info
Could not connect to host 192.168.1.238:9999
pi@octopi3:~/tplink-smartplug $ sudo ./tplink_smartplug.py -t 192.168.1.238 -c info
Could not connect to host 192.168.1.238:9999
`

MatchstickWorks, 2022-12-16:
> There are two ways we can try to resolve this error @MatchstickWorks. Either manually deleting the database over SSH and restarting OctoPrint, or clearing the settings from Plugin Manager completely [...]

That seemed to do it for me! Thanks for the help!

jneilliii, 2022-12-16:
> i did some testing with the original communication script, looks like port 9999 no longer works....

That's unfortunate. Seems maybe your plug updated with the new connection protocol. Found a similar post with additional links [here](https://github.com/python-kasa/python-kasa/issues/115).

> That seemed to do it for me! Thanks for the help!

Great to hear.

jneilliii, 2022-12-16:
Curious if you get the same thing with the python-kasa library @kevman28? If you run the following does it return anything? This module is what Home Assistant uses and I could potentially re-program my plugin around that library (would take a while to complete this task). 

```
source /home/pi/oprint/bin/activate
pip install python-kasa
kasa --host 192.168.1.220
```

kevman28, 2022-12-16:
I think it still tries port 9999, but also seeing some script errors

`(oprint) pi@octopi2:~ $ kasa --host 192.168.1.238
No --type defined, discovering..
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/protocol.py", line 131, in _query
    await self._connect(timeout)
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/protocol.py", line 81, in _connect
    self.reader, self.writer = await asyncio.wait_for(task, timeout=timeout)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
  File "/usr/lib/python3.7/asyncio/streams.py", line 77, in open_connection
    lambda: protocol, host, port, **kwds)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 959, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.7/asyncio/base_events.py", line 946, in create_connection
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 464, in sock_connect
    return await fut
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 494, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.1.238', 9999)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/bin/kasa", line 8, in <module>
    sys.exit(cli())
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 1157, in __call__
    return anyio.run(self._main, main, args, kwargs, **opts)
  File "/home/pi/oprint/lib/python3.7/site-packages/anyio/_core/_eventloop.py", line 70, in run
    return asynclib.run(func, *args, **backend_options)
  File "/home/pi/oprint/lib/python3.7/site-packages/anyio/_backends/_asyncio.py", line 292, in run
    return native_run(wrapper(), debug=debug)
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/oprint/lib/python3.7/site-packages/anyio/_backends/_asyncio.py", line 287, in wrapper
    return await func(*args)
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 1160, in _main
    return await main(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 1076, in main
    rv = await self.invoke(ctx)
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 1665, in invoke
    rv = await super().invoke(ctx)
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 1434, in invoke
    return await ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/lib/python3.7/site-packages/asyncclick/core.py", line 780, in invoke
    rv = await rv
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/cli.py", line 98, in cli
    dev = await Discover.discover_single(host)
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/discover.py", line 204, in discover_single
    info = await protocol.query(Discover.DISCOVERY_QUERY)
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/protocol.py", line 73, in query
    return await self._query(request, retry_count, timeout)
  File "/home/pi/oprint/lib/python3.7/site-packages/kasa/protocol.py", line 141, in _query
    f"Unable to connect to the device: {self.host}: {ex}"
kasa.exceptions.SmartDeviceException: Unable to connect to the device: 192.168.1.238: [Errno 113] Connect call failed ('192.168.1.238', 9999)
`

jneilliii, 2022-12-16:
Yeah, looks like you're right. It's the 9999 port not accepting connections still. Just curious, have you verified your IP address? Unplugged the device from the wall and plugged it back in to reboot it?

kevman28, 2022-12-17:
so I did some more troubleshooting....it looks like this might be an issue in my network, I have one Pi that is able to connect and run those queries to the switches, but three others that can't. I am stumped at this point, may consider rebuilding those octopi systems...its basically unable to ping those devices on the the systems that are not working...

kevman28, 2022-12-17:
i think I finally figured out this problem it is not related to the script or plugin at all. My asus router has a setting called "Target Wake Time" , I disabled it and all the devices started working again. Also as per TP link, my HS103 devices still have the open API, they did not update those to block port 9999. Hope this is helpful for anyone else here. Thank you jneilliii for all the help and support. Many thanks !!

jneilliii, 2022-12-18:
Thanks for the tip. I've actually seen that setting before relative to octopi devices going offline.

closed 2022-12-18 by jneilliii