jneilliii/OctoPrint-TPLinkSmartplug #325: [BUG]: getAdditionalData hook overrides other plugin

larsjuhw, 2022-12-09:
Hi,
This plugin [overwrites](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/c69a0ee89a239d9b44f5d48572bcba9ea165131c/octoprint_tplinksmartplug/static/js/tplinksmartplug.js#L15) the filesViewModel.getAdditionalData function to display the power cost of the last print in the file information view. However, my plugin ExtraFileInfo [also hooks](https://github.com/larsjuhw/OctoPrint-ExtraFileInfo/blob/9c72498698c873eb5b10bb1c378fba75a10c8c6d/octoprint_extrafileinfo/static/js/extrafileinfo.js#L67) that same function meaning the plugins can not work together as described here https://github.com/larsjuhw/OctoPrint-ExtraFileInfo/issues/12. The ExtraFileInfo plugin loads first meaning TPLinkSmartplug overrides the hook.

Could you consider appending to the existing getAdditionalData function instead of overwriting it, so that our plugins do not conflict?

jneilliii, 2022-12-09:
I'd be more than happy to work through this with you. Is that how your plugin is doing it? I can look at its implementation and adjust mine the same if that is the case.

larsjuhw, 2022-12-10:
Wow that was quick. Thank you very much!

closed 2023-03-18 by jneilliii