jneilliii/OctoPrint-TPLinkSmartplug #326: [Feature Request]: Save used Energy Data to Print

Andy2203, 2022-12-28:
Hello,

a cool plugin that I have been using for a long time.

I have 2 questions:
Where do I set the value to get the cost of consumption angeziegt?
There is a graph for cost in the chart, but it is always at 0.
[image]

Is it possible to assign the cost and the kWh consumed in a history to the print job? Or similar to the file manager "last print duration".
[image]

jneilliii, 2022-12-28:
It actually will show up in the file list if you enter the cost in settings.

[image]

Andy2203, 2022-12-28:
Thx, i did not see it.
What about a historie for single print jobs?

jneilliii, 2022-12-28:
it will only store from the last print when that value is set, so no history and no previous printed files will be able to be updated.