jneilliii/OctoPrint-TPLinkSmartplug #329: [BUG]: HS300 outlet index is 1-based, not 0-based

timur-tabi, 2023-03-17:
I just bought a new HS300 and updated the firmware.  It has six outlets numbered 1 though 6.  My printer is plugged into outlet 2 and my lights are plugged into outlet 3.

According to the documentation, I should specify 192.168.1.22/1 to switch my printer and 192.168.1.22/2 to switch the lights.  However, I actually need to use 192.168.1.22/2 to switch the printer, and 192.168.1.22/3 to switch the lights.

Octoprint 1.8.7

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/10997102/plugin_tplinksmartplug_debug.log)

Note that the log shows other IP addresses.  I bought this outlet to replace two individual HS100 outlets.

jneilliii, 2023-03-17:
This is a documentation error. There was a version released that changed the logic here and matched up the numbers to what's listed on the strip.

jneilliii, 2023-03-17:
the docs have actually already been updated in the rc branch, and will become live when the next version is released.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/tree/rc

timur-tabi, 2023-03-17:
FYI, the link https://www.tp-link.com/us/support/faq/2707/ in the readme is no longer valid.

jneilliii, 2023-03-18:
Yeah, looks like TPLink deleted that FAQ. Luckily, we have the internet time machine...

http://web.archive.org/web/20220707141756/https://www.tp-link.com/us/support/faq/2707/

jneilliii, 2023-03-18:
added pdf version of the FAQ and updated the readme, just released 1.0.3.

closed 2023-03-18 by jneilliii

timur-tabi, 2023-03-18:
I'm not sure those instructions are valid anymore.  The "continue without creating an account" option is not present in the app, and least not on my phone.

jneilliii, 2023-03-18:
It's also dependent on which device, the HS300 requires remote connection but the HS110 didn't.

[image]

timur-tabi, 2023-03-18:
That explains it.  Oh well.  At least I was able to use https://www.fakenamegenerator.com/ to avoid giving my real email address.