jneilliii/OctoPrint-TPLinkSmartplug #331: [Feature Request]:

alessio-pugliese, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
I'm running octoprint via a docker container which has its specific network, as per the default config (see [here](https://hub.docker.com/r/octoprint/octoprint)). If I ping from within a container terminal shell the IP address of my TPLink P110, which lives in my home network, I can successfully exchange packets of data, this means that the container see the smart plug is up.  The problem is when I put the IP address of the smart plug in the TPlink plugin config, it doesn't seem to hook up to the smart plug.  

**Describe the solution you'd like**
The solution request is pretty straightforward. I'd like to have a solution for those who runs octoprint in a docker container. 

**Describe alternatives you've considered**
The alternative is to config the docker octoprint container to run on the host network. In this case network isolation is lost though.

alessio-pugliese, 2023-03-24:
I can confirm that my tapo p110 smart plug works flawlessly with PSUControl plugin combined with PSUControl -tapo subplugin. Sorry if I bothered with docker nonsense. It's a shame we Tapoers cannot give a try to your plugin though. 
Thanks for the support.

jneilliii, 2023-03-24:
There is a fork of this plugin for tapo devices, but since I don't have any don't know how well it works.

https://github.com/Tyfy/OctoPrint-TapoSmartplug

closed 2023-03-24 by jneilliii

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.