jneilliii/OctoPrint-TPLinkSmartplug #332: With KP303 and lamp on port 2 kills connection to printer

Grayzone233, 2023-03-25:
Using KP303 switch lamp on port 2 Optoprint on port 1. If you have a connection from Octoprint to the printer and turn off the Lamp, Octoprint will disconnect from the printer.  This is with the latest 1.0.3 version, the earlier version worked fine. Not a good thing when you are printing some thing as it kills it.

**To Reproduce**
Steps to reproduce the behavior:
1. web interface Octoprint, connect to printer and have lamp on.
2.3. Click on the lamp power button to turn of lamp.
4. Printer is now disconnected,

**Expected behavior**
Expect the printer to stay connected.

**Desktop (please complete the following information):**
 - OS: Windows 10
 -  - Browser Google Chrome
 -  - OctoPrint Version 1.8.7
 - Plugin Version 1.0.3

jneilliii, 2023-03-25:
Please enable debug logging in the plugins settings, restart OctoPrint and reproduce the issue. Then share the plugin_tplinksmartpkug_debug.log file from OctoPrint's logging section in settings and screenshots of all your settings for the plugin.

Grayzone233, 2023-03-25:
Here you go.
[image]
[image]
[octoprint.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/11069931/octoprint.zip)

jneilliii, 2023-03-25:
Please provide the individual plug settings as well and that zip doesn't seem to contain the `plugin_tplinksmartpkug_debug.log` file requested, just octoprint.log (which has loads of smartfilamentsensor plugin errors BTW).

Grayzone233, 2023-03-25:
Ok, last time it did not create a tplin...log file. Here are the two plug settings (everything below the cut off is off.
Trying to get log file now.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/11070343/plugin_tplinksmartplug_debug.log)

[image]
[image]

jneilliii, 2023-03-25:
Disable the Auto Disconnect and Auto Connect options on the Lamp settings. You could also probably disable the warning prompt and warn while printing for that one as well.

Grayzone233, 2023-03-26:
Thanks that fixed it!!!

closed 2023-03-26 by Grayzone233