jneilliii/OctoPrint-TPLinkSmartplug #333: Plug set for LED light does not turn on when the printer turns on...

rashuge, 2023-04-02:
I have my printer setup to turn on when I send a job to OctoPrint.  I have that configured in Cura.  There is no setting for another plug but I have the plug for the LED lights configured properly I believe.  It does not actually turn on when the printer turns on.  Looking for some guidance on the issue.  Below is the screen shot of my configuration.  Any recommendations or feedback would be appreciated.

[image]

jneilliii, 2023-04-02:
would need to see your settings for the LED Lights, but my initial though would be to enable gcode triggers for those, and then in OctoPrint's start print or on connect GCODE settings add in the `@TPLINKON 192.168.0.2` trigger.

jneilliii, 2023-04-03:
Yeah, I would disable the auto connect and auto disconnect options (that can cause you headaches later if you try to turn off light while printing), enable gcode triggers, and then in OctoPrint's gcode scripts settings for on printer connection add teh appropriate gcode trigger for the light. That way the Cura plugin will power on the printer and connect to the printer, the on connection script will then send that command and be picked up by the plugin and power on your light.

closed 2023-04-25 by github-actions[bot]