jneilliii/OctoPrint-TPLinkSmartplug #335: [BUG]: Delayed GCODE off should be ignored if GCODE on sent

ttabi, 2023-04-15:
I have the following in my custom GCODE:

@TPLINKON 192.168.1.22/3 ; Lights
...
@TPLINKOFF 192.168.1.22/3

That is, the lights turn on when the print starts, and they turn off when the print stops.  However, I have "GCODE Off Delay" set to 60 seconds.  60 seconds after the print stops, the lights turn off.  So far so good.

However, if i start another print (with the same GCODE commands) within 60 seconds, the lights will still turn off even though a new print has started.  That is because the 60-second delay for "@TPLINKOFF 192.168.1.22/3" is still running, even though I already sent a new "@TPLINKON 192.168.1.22/3". 

So when there is a delayed TPLINKOFF still waiting, the plugin should delete that TPLINKOFF timer if it gets a matching TPLINKON before the delay expires.

My Python passable, but I have to admit the plugin code is a bit confusing for me.  However, I believe the fix is to change tplinksmartplugPlugin::turn_on() so that it checks if there is an existing turn_off() timer running, and if so, kill the timer.

jneilliii, 2023-04-15:
> My Python passable, but I have to admit the plugin code is a bit confusing for me. However, I believe the fix is to change tplinksmartplugPlugin::turn_on() so that it checks if there is an existing turn_off() timer running, and if so, kill the timer.

you followed it proper, that would be the correct way to counter this issue. thanks for the report.

jneilliii, 2023-04-15:
The only caveat is that it should only cancel an off timer for the same ip and not all of them, although now that I'm looking at the code, this will probably take a good amount of reprogramming to make those timers accessible outside of the processAtCommand function.

jneilliii, 2023-04-16:
A workaround that you can use to get around this issue for now is if you enable warn while printing option on the light and a print has started it won't trigger the off command.

jneilliii, 2023-04-16:
added timer tracking in the above commit and is available as version 1.0.4rc1. change the release channel for the plugin in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

closed 2025-08-13 by jneilliii