jneilliii/OctoPrint-TPLinkSmartplug #336: [BUG]: With Prusa MK4 connected , the plugin fails to detect idle state and doesnt turn off the plug.

kevman28, 2023-06-02:
**Describe the bug**
With Prusa MK4 connected , the plugin fails to detect idle state and doesn't turn off the plug.

jneilliii, 2023-06-02:
Please share screenshots of your overall settings and the individual plug's settings. Typically this is because you didn't enable the plug settings to turn off on idle.

kevman28, 2023-06-02:
> Please share screenshots of your overall settings and the individual plug's settings. Typically this is because you didn't enable the plug settings to turn off on idle.

[image]
[image]

jneilliii, 2023-06-03:
great, wasn't what I thought. If you don't mind, I see you have debug logging enabled and as long as OctoPrint was restarted after enabling that you should see a file plugin_tplinksmartplug_debug.log in the logging section of OctoPrint's settings. just to double-check use the icon to power off/on the printer and after the amount of time has elapsed for the idle timeout to trigger, let's see what's in that file.

kevman28, 2023-06-03:
[2023-06-03 08:12:18,304] DEBUG: TIM
[2023-06-03 08:13:17,139] DEBUG: Turning on 192.168.1.196/3.
[2023-06-03 08:13:17,141] DEBUG: {'autoConnect': True, 'autoConnectDelay': '14', 'autoDisconnect': True, 'autoDisconnectDelay': 0, 'automaticShutdownEnabled': True, 'btnColor': '#808080', 'countdownOffDelay': 1, 'countdownOnDelay': 1, 'currentState': 'off', 'displayWarning': True, 'emeter': {'get_realtime': {}}, 'event_on_disconnect': False, 'event_on_error': False, 'event_on_shutdown': False, 'event_on_startup': False, 'event_on_upload': True, 'gcodeCmdOff': False, 'gcodeCmdOn': False, 'gcodeEnabled': False, 'gcodeOffDelay': 0, 'gcodeOnDelay': 0, 'gcodeRunCmdOff': '', 'gcodeRunCmdOn': '', 'icon': 'icon-bolt', 'ip': '192.168.1.196/3', 'label': 'Mk4', 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'thermal_runaway': False, 'useCountdownRules': False, 'warnPrinting': False}
[2023-06-03 08:13:17,142] DEBUG: IP 192.168.1.196 is valid.
[2023-06-03 08:13:17,156] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-06-03 08:13:17,157] DEBUG: IP 192.168.1.196 is valid.
[2023-06-03 08:13:17,158] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.196
{"id":"2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}}],"child_num":3,"ntc_state":0,"err_code":0}}}
[2023-06-03 08:13:17,209] DEBUG: Sending command {'system': {'set_relay_state': {'state': 1}}, 'context': {'child_ids': ]}} to 192.168.1.196
[2023-06-03 08:13:17,230] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2023-06-03 08:13:17,230] DEBUG: 0
[2023-06-03 08:13:17,230] DEBUG: Resetting idle timer since plug 192.168.1.196/3 was just turned on.
[2023-06-03 08:13:17,231] DEBUG: Checking status of 192.168.1.196/3.
[2023-06-03 08:13:17,231] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-06-03 08:13:17,231] DEBUG: IP 192.168.1.196 is valid.
[2023-06-03 08:13:17,237] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-06-03 08:13:17,237] DEBUG: IP 192.168.1.196 is valid.
[2023-06-03 08:13:17,237] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.196

[2023-06-03 08:13:17,314] DEBUG: TIM
[2023-06-03 08:14:31,495] DEBUG: Idle timeout reached after 1 minute(s). Turning heaters off prior to powering off plugs.
[2023-06-03 08:14:31,496] DEBUG: Heater tool0 already off.
[2023-06-03 08:14:31,496] DEBUG: Heater bed already off.
[2023-06-03 08:14:31,496] DEBUG: Turning off heater: X

kevman28, 2023-06-03:
[2023-06-03 08:14:31,496] DEBUG: Turning off heater: X
[2023-06-03 08:20:52,238] DEBUG: resetting idle timer due to ClientOpened event

looks like right here it resets after a few mins...

jneilliii, 2023-06-03:
> Turning off heater: X
This is a new one. If you go to your terminal tab in OctoPrint, please share a temperature report line. It appears they may have added something that option is trying to turn off and can't maybe.

The rest issue timer happens when you connect to the web interface, so that's kind of expected.

kevman28, 2023-06-04:
Recv: FIRMWARE_NAME:Prusa-Firmware-Buddy 5.0.0-ALPHA1+11007 (Github) SOURCE_CODE_URL:https://github.com/prusa3d/Prusa-Firmware-Buddy PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa-mini EXTRUDER_COUNT:5 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Recv: ok
Send: M155 S2
Recv: ok
Recv:  T:25.00/0.00 B:24.49/0.00 X:24.72/36.00 A:26.51/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.49/0.00 X:24.70/36.00 A:26.64/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.72/36.00 A:26.80/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.19/0.00 B:24.61/0.00 X:24.70/36.00 A:26.98/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.73/36.00 A:27.16/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.72/36.00 A:27.19/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.55/0.00 X:24.71/36.00 A:27.45/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.74/36.00 A:27.50/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.49/0.00 X:24.71/36.00 A:27.60/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.37/0.00 X:24.71/36.00 A:27.71/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.40/0.00 X:24.70/36.00 A:27.97/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.43/0.00 X:24.70/36.00 A:28.02/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.71/36.00 A:28.07/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.72/36.00 A:28.10/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.74/36.00 A:28.23/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.72/36.00 A:28.39/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.72/36.00 A:28.52/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.72/36.00 A:28.57/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.71/36.00 A:28.72/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.71/36.00 A:28.80/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.71/36.00 A:28.83/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.58/0.00 X:24.74/36.00 A:28.96/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.67/0.00 X:24.72/36.00 A:29.01/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.75/0.00 B:24.58/0.00 X:24.73/36.00 A:29.19/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.73/36.00 A:29.11/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.73/36.00 A:29.22/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.74/36.00 A:29.27/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.43/0.00 X:24.73/36.00 A:29.43/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.61/0.00 X:24.74/36.00 A:29.51/0.00 @:0 B@:0 HBR@:0
Recv:  T:26.22/0.00 B:24.64/0.00 X:24.73/36.00 A:29.56/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.73/36.00 A:29.66/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.61/0.00 X:24.73/36.00 A:29.66/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.67/0.00 X:24.74/36.00 A:29.79/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.47/0.00 B:24.49/0.00 X:24.73/36.00 A:29.84/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.74/36.00 A:29.92/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.52/0.00 X:24.74/36.00 A:29.95/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.47/0.00 B:24.64/0.00 X:24.75/36.00 A:30.04/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.47/0.00 B:24.55/0.00 X:24.73/36.00 A:30.04/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.73/36.00 A:30.02/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.19/0.00 B:24.55/0.00 X:24.74/36.00 A:30.17/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.70/36.00 A:30.15/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.19/0.00 B:24.49/0.00 X:24.72/36.00 A:30.21/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.73/36.00 A:30.27/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.75/36.00 A:30.31/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.56/0.00 B:24.58/0.00 X:24.75/36.00 A:30.40/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.73/36.00 A:30.35/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.75/0.00 B:24.52/0.00 X:24.73/36.00 A:30.44/0.00 @:0 B@:0 HBR@:0
Recv:  T:26.03/0.00 B:24.52/0.00 X:24.74/36.00 A:30.48/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.75/36.00 A:30.56/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.55/0.00 X:24.74/36.00 A:30.62/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.19/0.00 B:24.55/0.00 X:24.75/36.00 A:30.62/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.74/36.00 A:30.71/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.58/0.00 X:24.74/36.00 A:30.69/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.09/0.00 B:24.58/0.00 X:24.75/36.00 A:30.75/0.00 @:0 B@:0 HBR@:0
Recv:  T:25.00/0.00 B:24.52/0.00 X:24.73/36.00 A:30.81/0.00 @:0 B@:0 HBR@:0
Sen

jneilliii, 2023-06-06:
Do you happen to know what X is in that report?

jneilliii, 2023-06-06:
and A for that matter.

jneilliii, 2023-06-06:
Yeah, it's new and the idle timer keeps getting reset because the "target" is never able to be turned off. @kantlivelong will probably want to know about this as well, as I borrowed that idle timer code from PSU Control.

jneilliii, 2023-07-01:
re-opening this request @kevman28 as it is possible for me to work around this issue, just odd that Prusa is sending those additional temperatures.

jneilliii, 2023-07-01:
I've added a setting in the latest 1.0.4rc2 release that allows for setting heaters to be ignored coming from that temperature report. You should be able to add `X` to that setting to allow it to be ignored and the idle timeout work again. 

[image]

You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted to get the change and validate it's working again with the above setting.

If it does work, please report back here and to the Prusa Forum post about OctoPrint compatibility that I saw you mentioning this issue as well.

kevman28, 2023-07-02:
Thank you! I will test this out today and report back.

jneilliii, 2023-07-02:
please share updated log after upgrading to debug why it's still not working as anticipated.

kevman28, 2023-07-02:
I'm happy to report that this is resolved completely with the MK4 in the release candidate. Thank you for your great efforts here.

closed 2023-07-02 by jneilliii

ScottWell1, 2023-07-11:
I know this is closed but wanted to add some info about the MK4 temperature report format in case someone comes across it when searching.

The "X:" is the heatbreak temperature (actual / target).  Heatbreak target is set with "M142 Snn".  The firmware controls the heatbreak fan speed in an attempt to keep it at or below the target temp.

The "A:" is the controller board (MK4 "buddy board") temperature.  If Prusa followed the same design as for their MINI model, the thermistor is located adjacent to the high-load connection for the heated bed - but I do not have confirmation of that yet.  Only actual temp is reported (target is always 0).

timur-tabi, 2023-11-06:
I think this problem might affect MK3 as well.  My printer also no longer turns off on idle.  Here's what the terminal outputs:

Recv: T:28.1 /0.0 B:27.9 /0.0 T0:28.1 /0.0 @:0 B@:0 P:28.8 A:34.8
Recv: T:27.9 /0.0 B:27.9 /0.0 T0:27.9 /0.0 @:0 B@:0 P:29.0 A:33.9
Recv: T:27.9 /0.0 B:27.6 /0.0 T0:27.9 /0.0 @:0 B@:0 P:29.0 A:34.4
Recv: T:28.0 /0.0 B:28.0 /0.0 T0:28.0 /0.0 @:0 B@:0 P:29.1 A:34.6

jneilliii, 2023-11-06:
Looks like you will probably need the release candidate as well and add ignoring the `P,A` to the ignore heaters line.

timur-tabi, 2023-11-06:
I take that back.  My printer just turned off by itself.  I guess I just didn't wait as long as I thought I did.