jneilliii/OctoPrint-TPLinkSmartplug #337: [BUG]: Tp link plug won't connect

Frankiemagic13, 2023-10-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Plugged the plug in and registered the ip address it's a static IP and it won't connect 

**Expected behavior**
Trigger the plug through octoprint 
**Screenshots**
[image]
**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser Chrome
 - OctoPrint Version 1.9.0
 - Plugin Version
1.0.3
**Additional context**
Add any other context about the problem here.

jneilliii, 2023-10-08:
what type of plug and which version firmware? the plugin sees the IP and sends the command to it for status information and the plug is not allowing the connection or not responding.

Frankiemagic13, 2023-10-08:
> what type of plug and which version firmware? the plugin sees the IP and sends the command to it for status information and the plug is not allowing the connection or not responding. 

I have a HS105 version 1.5.6 the weird thing is it's worked for a while then out of no where stopped working

Frankiemagic13, 2023-10-08:
> what type of plug and which version firmware? the plugin sees the IP and sends the command to it for status information and the plug is not allowing the connection or not responding. 

Anything I can test on my end ?

jneilliii, 2023-10-08:
The only thing I can think of you can try is to unplug the device from the wall completely and plug it back in. Verify from your router that it did get the same IP address as expected and see if it's any different. I don't have an HS105 on hand to test with, but my HS103 with firmware version 1.0.3 and 1.0.4 are still working as expected. It's possibly you got a firmware update that has now blocked communication. That has happened before for some UK users.

Frankiemagic13, 2023-10-08:
> users.

Is there a way to force an older update firmware

jneilliii, 2023-10-08:
That I am unsure of.

Frankiemagic13, 2023-10-08:
> The only thing I can think of you can try is to unplug the device from the wall completely and plug it back in. Verify from your router that it did get the same IP address as expected and see if [...]

Found the new IP address but still no luck here's the log 

[2023-10-08 15:18:06,027] DEBUG: 
[2023-10-08 15:18:06,028] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2023-10-08 15:19:24,658] DEBUG: abortTimeout: 30
[2023-10-08 15:19:24,662] DEBUG: powerOffWhenIdle: True
[2023-10-08 15:19:24,667] DEBUG: idleTimeout: 30
[2023-10-08 15:19:24,671] DEBUG: idleIgnoreCommands: M105
[2023-10-08 15:19:24,674] DEBUG: idleTimeoutWaitTemp: 50
[2023-10-08 15:19:51,825] DEBUG: resetting idle timer due to ClientOpened event.
[2023-10-08 15:19:57,863] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.7.232')])
[2023-10-08 15:19:57,863] DEBUG: Checking status of 192.168.7.232.
[2023-10-08 15:19:57,864] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-10-08 15:19:57,864] DEBUG: IP 192.168.7.232 is valid.
[2023-10-08 15:19:57,865] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.7.232
[2023-10-08 15:20:27,891] DEBUG: Could not connect to 192.168.7.232.
[2023-10-08 15:20:27,892] DEBUG: 
[2023-10-08 15:20:27,893] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2023-10-08 15:21:55,636] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.7.232')])
[2023-10-08 15:21:55,637] DEBUG: Checking status of 192.168.7.232.
[2023-10-08 15:21:55,638] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-10-08 15:21:55,638] DEBUG: IP 192.168.7.232 is valid.
[2023-10-08 15:21:55,638] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.7.232
[2023-10-08 15:22:25,662] DEBUG: Could not connect to 192.168.7.232.
[2023-10-08 15:22:25,663] DEBUG: 
[2023-10-08 15:22:25,664] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}
[2023-10-08 15:22:25,825] DEBUG: Settings saved, Automatic Power Off Enabled, starting idle timer...
[2023-10-08 15:22:45,605] DEBUG: ImmutableMultiDict([('checkStatus', '192.168.7.232')])
[2023-10-08 15:22:45,605] DEBUG: Checking status of 192.168.7.232.
[2023-10-08 15:22:45,606] DEBUG: {'system': {'get_sysinfo': {}}}
[2023-10-08 15:22:45,606] DEBUG: IP 192.168.7.232 is valid.
[2023-10-08 15:22:45,606] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.7.232
[2023-10-08 15:23:15,636] DEBUG: Could not connect to 192.168.7.232.
[2023-10-08 15:23:15,637] DEBUG: 
[2023-10-08 15:23:15,638] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}, 'emeter': {'err_code': True}}

Frankiemagic13, 2023-10-08:
> The only thing I can think of you can try is to unplug the device from the wall completely and plug it back in. Verify from your router that it did get the same IP address as expected and see if [...]

The bolt shows up grey[image]

jneilliii, 2023-10-09:
Yeah, so same thing. The plug is not taking the status command, which in turn is why the icon stays gray. You don't happen to see any errors in octoprint.log related to the plugin?

Frankiemagic13, 2023-10-09:
> Yeah, so same thing. The plug is not taking the status command, which in turn is why the icon stays gray. You don't happen to see any errors in octoprint.log related to the plugin? 

Let me take a look

Frankiemagic13, 2023-10-09:
> Yeah, so same thing. The plug is not taking the status command, which in turn is why the icon stays gray. You don't happen to see any errors in octoprint.log related to the plugin?
[octoprint.pdf](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/12842510/octoprint.pdf)

heres the log

Frankiemagic13, 2023-10-09:
> Yeah, so same thing. The plug is not taking the status command, which in turn is why the icon stays gray. You don't happen to see any errors in octoprint.log related to the plugin? 

I saw a exclamation point next to the tp link plugin idk if there's something wrong on that log I sent

jneilliii, 2023-10-09:
That log is so full of Obico errors it's hard to filter through aa PDF. You should be able to drag the log file directly into a comment here to upload. Initial glance I don't see anything though outside a ton of Obico errors.

Frankiemagic13, 2023-10-12:
here is the log sorry for delay in response 
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/12876326/octoprint.log)

Frankiemagic13, 2023-10-12:
> That log is so full of Obico errors it's hard to filter through aa PDF. You should be able to drag the log file directly into a comment here to upload. Initial glance I don't see anything though outside a ton of Obico errors.

[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/12876327/octoprint.log).  Here it is sorry for the delay

jneilliii, 2023-10-12:
Yeah, definitely nothing in there that helps in debugging why it just stopped working all of a sudden.

Frankiemagic13, 2023-10-12:
> working

So what should I try next ?

Frankiemagic13, 2023-10-12:
> Yeah, definitely nothing in there that helps in debugging why it just stopped working all of a sudden. 

Do I need to be on the same network the plug is connected too?

jneilliii, 2023-10-12:
The pi needs to be on the same network, or at least be able to properly route traffic between networks, but they have to be able to communicate with each other for sure.

Frankiemagic13, 2023-10-12:
> The pi needs to be on the same network, or at least be able to properly route traffic between networks, but they have to be able to communicate with each other for sure. 

Ok let me try switching networks for the plug

Frankiemagic13, 2023-10-12:
> The pi needs to be on the same network, or at least be able to properly route traffic between networks, but they have to be able to communicate with each other for sure. 

So update for some reason my raspberry pi was on a different network that the plug couldn't connect to due to wpa protocol so I restored the pi to a different network that the plug was able to connect too and it all works now :D sorry for all the annoyance

jneilliii, 2023-10-12:
no worries, this has happened before to others and will surely happen again in the future.

closed 2023-10-12 by jneilliii