jneilliii/OctoPrint-TPLinkSmartplug #338: [BUG]:

amweiss, 2023-11-26:
**Describe the bug**
Failure on startup. This occurred after a brownout so a few devices and wifi signal might have been unavailable.

```
2023-11-26 07:37:01,386 - octoprint.plugin - ERROR - Error while calling plugin tplinksmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugin/__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_tplinksmartplug/__init__.py", line 633, in on_event
    response = self.turn_on(plug["ip"])
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_tplinksmartplug/__init__.py", line 415, in turn_on
    return self.check_status(plugip)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_tplinksmartplug/__init__.py", line 473, in check_status
    timer_chk = self.lookup(response, *["system", "get_sysinfo", "children"])[int(plug_ip[1]) - 1][
TypeError: 'NoneType' object is not subscriptable
```

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on raspberry pi power
2. See error in logs

**Expected behavior**
Startup occurs without exception.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Octopi 1.0.0 Build 2023.02.22.095002
 - OctoPrint Version 1.9.3
 - Plugin Version 1.0.3

**Additional context**
N/A

jneilliii, 2023-11-26:
So I assume you have startup event monitoring and that at least a single plug has this checkbox checked? The error you're getting makes me think you might be missing the individual pliug's checkbox. 

[image]

amweiss, 2023-11-27:
Yep, it's on and it normally works fine but this one time when starting it failed.

[image]

jneilliii, 2023-11-28:
Could be some random race condition maybe that caused it. The error seems to indicate that it may not have gotten a response from the plug for it's initial status check, which caused the error. Not sure if there's going to be much I can do to help in this situation since you mentioned it normally works unless we can figure out some common denominator when it doesn't.

closed 2023-12-21 by github-actions[bot]