jneilliii/OctoPrint-TPLinkSmartplug #341: [BUG]: No longer working

narfotic, 2024-01-09:
**Describe the bug**
I can no longer control or see the status of my plug. It shows a grey icon in my header.
I can control the plug from the Kasa app.
I doublechecked that the IP-address was correct.
I un- and reinstalled the plugin.

**Expected behavior**
I would like to be able to control the plug via the plugin.

I have a TP Link HS110 (EU version)  - HW version 4.0 - Firmware version 1.0.5
Running Octoprint Version 1.9.3
Octopi Version 0.18.0, running on Raspberry Pi Zero 2 Rev 1.0

jneilliii, 2024-01-09:
If the IP is correct then that points to either a firmware or network related issue with the communication between the pi and the plug. Your description makes me think it worked at some point? You can enable debug logging in the plugin's settings, restart OctoPrint and then share the file plugin_tplinksmartplug_debug.log from OctoPrint's logging section in settings for further troubleshooting. Make sure you click the gray icon a couple of times to try and gleam current status. You can also try unplugging the tplink from the wall for a little bit and plugging in back in (make sure the IP doesn't change on you).

narfotic, 2024-01-12:
I did work great before!
Since some time it stopped working. I thought it must be some IP related thing, so I never tried debugging till now. So cannot say exactly when it stopped working.

I have double checked the IP even changed it in both my router and the plugin to see if that would change anything.
I unplugged it for a day.
From the log it clearly shows DEBUG: "Could not connect to 192.168.1.234"
[plugin_tplinksmartplug_debug (2).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/13919817/plugin_tplinksmartplug_debug.2.log)

jneilliii, 2024-01-12:
Did anything else change on the network or the device? The only other time I've seen this be an issue was either new firmware blocking connections or a routing problem between the pi and plug. In the case of the routing problem it was because the device was on an isolated network/VLAN.

narfotic, 2024-01-14:
Thank you for setting me on the right path! I did not change anything on purpose but there must be some settings which did not allow communication between my regular and my IOT wifi. Moving my octopi to my IOT network solved the problem! Thanks again and sorry to bother you with my lack of networking knowledge.

closed 2024-01-14 by narfotic