jneilliii/OctoPrint-TPLinkSmartplug #344: [Feature Request]: Add additional field to database representing "corrected" total energy usage

puterboy, 2024-02-04:
I noticed that sometimes my TPLink plug seems to reset (maybe do to a firmware update or just a glitch), in which case the `total` column in the `energy_data.db` database reverts to 'zero'
This of course messes up the cumulative energy consumption and shows up as a sharp drop in the total in the corresponding graph.

This can be corrected by adding an additional `corrected_total` field to the database, calculated as follows:
- If new `total` >= old `total`, `corrected_total` = `total`
- else `corrected_total` = old `total` + `total`

Now this might night catch all reset situations if the plug resets and then runs for a long time without octoprint running so that the reset total exceeds the last stored total, but it should correct most errors...

Adding this correction should help with longer term energy tracking...
And in any case you would still have the raw 'total' field present...

You may also want to add a reset button that would allow one to manually reset the `total'

jneilliii, 2024-02-04:
firmware update or factory reset are the only ways I can think of that would cause the total to reset.

puterboy, 2024-02-04:
That's what I assumed but somehow on Jan 7 this year the plug twice reset to zero. I don't think I was playing with it but you never know... :)
Still I think it would be nice to keep a running monotonic energy total as there are valid reasons to update firmware, reset plugs (e.g., if changing SSID or password or just getting a new plug)

puterboy, 2024-03-03:
Just noticed that plug reset on about Feb 29th -- even though (or maybe because) I hadn't used it in 2 weeks.
Could be an automatic firmware update or could be something in the existing firmware that triggers reset.
Either way the PR I posted fixes this..

closed 2025-08-13 by jneilliii