jneilliii/OctoPrint-TPLinkSmartplug #345: [Feature Request]: Only add entries to db if current or power non-zero or if total changes

puterboy, 2024-02-04:
The `energy_data.db` continues to grow with new entries even if `current=0`, `power=0` and `total` is unchanged.
The only thing of note being recorded is the `voltage` which is not very relevant to Octoprint in general.

To avoid database bloat when printer is not even on but Octoprint is running, I would suggest that such unchanged entries not be added to the database.

Not a big deal... just a suggestion to avoid unnecessary data storage...

jneilliii, 2024-02-04:
Don't poll status all the time and only poll during print progress and that will result in a less bloated db. I was curious and looked at my instance and I'm polling every minute, it was installed a very long time ago and still is only 720mb so feel this to be a fairly low priority. I am open to any PR that implements this change though.

puterboy, 2024-02-07:
Mine polls more frequently when printing but continues to poll exactly every 5 seconds even after printing and even when the TPLink switch (and printer) are off.
My file is 2MB after one month.

BTW, I think it's good to continue polling after printing since there is electricity usage if you just leave the printer and switch on -- and that is good to track. Just saying that one needn't record entries with no power change.

I can look at the code when I get a chance and will see what I can come up with...

puterboy, 2024-02-08:
I just made this and the other fixes to db on related issues and will share a PR soon.

One NICE benefit of only recording changes to the total is that the display for multi-date ranges is now MUCH faster since there are no points spanning down towns. This is really helpful since it used to take forever to load hundreds of thousands of points...

puterboy, 2024-02-08:
I liberalized the selection algorithm so it only records data when, EITHER:

- Power > 0 (so this captures all time power is being consumed
- Total changes (so this catches all increments in energy used)
- Power returns to 0 for the first time - this shows when the power actually turned off (and is nice for graphs I suppose)

closed 2025-08-13 by jneilliii