jneilliii/OctoPrint-TPLinkSmartplug #348: [BUG]: sending shutdown command to pi doesnt work. Log shows "could not connect.."

daveDGL, 2024-02-08:
As per octoprint community chat previously..

I have one KP105 plug that the plugin does as expected and shuts down the printer as per the settings.  Initially in that plug profile I added the before Off system command ""sleep 10s; shutdown -h now" but the log shows this is not sent.  I then added a second IP address/"plug" into the profile which is the IP address of the pi (192.168.0.100) just to issue the shutdown command to the Pi IP.  I can see in the logs that the plugin  first successfully shuts down the KP105 plug (192.168.0.99) but then cannot send the command to the Pi as it appears to be issuing TPLink specific commands that the Pi doesn't recognise. I've logged into the pi direct and tried the same commands ("system" and "get_sysinfo" and they are not recognised.

**Desktop (please complete the following information):**
 - OS: WIn11
 - Browser Google and Edge
 - OctoPrint Version 23.9.3 (old camera stack)
 - Plugin Version 1.0.3 from the octoprint log

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14207477/plugin_tplinksmartplug_debug.log)

daveDGL, 2024-02-08:
Hi, I deleted the fake plug and added the config you provided (screenshots of both pages of the config are attached).  I found the following:
1.  With the printer not powered on at all,  the plugin still activated  - the abort message appeared for the set 10secs then 65 seconds later the Pi shut down (seen via my vnc connection closing).  This is not good from my perspective as I often power up the pi to send files from cura for 'batch' printing later.
2. with the printer powered on via the plugin icon, but the printer left idle, the Pi shut down 5seconds after the abort timeout - this appears  the 5 seconds configured on page2.  I experimented with changing the 'timer off delay' setting on page 2 but it made no difference.  however changing the 'delay' setting did - whatever that setting was at, the pi shutdown that many seconds after the about message disappeared.  The printer plug then powered off 1 min later  - even when I changed the 'timer off delay' to 180 seconds and the idle timer to 2mins. The connection status of the printer in octoprint made no difference to this.

Given that the shutdown process works for both the pi and printer it doesn't really matter what order they shut down in (previously I thought the pi needed be running to shut down the TP link plug but now realise the command is issued correctly).  However, given that the plugin is shutting down pi even without the printer powered on, I have unchecked the run system code.

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/14213166/plugin_tplinksmartplug_debug.log)
[image]
[image]

jneilliii, 2024-02-08:
can you go into software update settings and change the release channel to `release candidate` for the plugin and update when prompted? I noticed that there were some changes specific to timers in one of those newer rc versions that might help. 

> 1. With the printer not powered on at all,  the plugin still activated  - the abort message appeared for the set 10secs then 65 seconds later the Pi shut down (seen via my vnc connection closing)

this has been a lingering thing I've been meaning to fix. that was one of the reasons I put the toggle switch for idle timer in the sidebar to easily turn it off when doing other stuffs like you mention.

closed 2024-03-01 by github-actions[bot]