jneilliii/OctoPrint-TPLinkSmartplug #349: [BUG]: 'Last power cost' in additional details is wrong

AndreasBoehme, 2024-02-08:
**Describe the bug**
The statistics of an already finished print shows wrong print-costs when using a TP-Link Smartplug and this OctoPrint-Plugin.

e.g. 
Last power cost: 6.305

TP-Link Smartplug Tab shows: 
total kWh: ranges from 1.06 to 2.393 => used: 1.333 kWh 
cost: ranges from 0.3392 to 0.76576 => 0,42656 Euro

(I checked that with the kasa-app from tp-link - the ptint was ~ 2 weeks ago, but the overall power consumtion is ~ 4 kWh, thus costs can't be higher than ~ 1.3 Euros)

**To Reproduce**
1. set Cost per kWh to '0.32'
2. start a long print (e.g. 15h)
3. show _TP-Link Smartplug_ Tab for the timespan of the print
4. when print is finished show additonal details of the print
5. _Last power Cost_ (from additional details) much higher than in the _TP-Link Smartplug_ Tab

**Additional context**

Seems the overall-power-costs is multiplied with the print-time.
I found this line in the source code: https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/a3771477d760215871bb98c8fb531fcbfb67f832/octoprint_tplinksmartplug/__init__.py#L707C22-L707C38

**Screenshots**
In the screenshot you see that the total power-consumption was 2.3 kWh when the print was finished.
The costs per kWh was configured to be 0.32.
Thus the _last power cost_ of 6.3 must be wrong (should be ~ 0.4)

[image]

[image]

jneilliii, 2024-02-08:
Can you do me a favor and enable debug logging in the plugin's settings, restart OctoPrint and go through a print again, then share the plugin_tplinksmartplug_debug.log file that will show up in OctoPrint's logging section of settings?

puterboy, 2024-02-09:
Agree it seems like a bug - I wrote some SQL queries to bring the Obico print history database plus the TPLink database into Excel pivot tables to calculate/display all sorts of stuff about my prints.
The energy costs that I calculate are a fraction of what the additional details shows...

puterboy, 2024-02-09:
I made the fix that you suggested -- i.e., remove the multiplication by `hours` and now the energy cost in Octoprint agrees with my database/spreadsheet/pivot table calculations.
@jneilliii I will include this in the PR I am testing for the other bugs/enhancements I have been working on...

AndreasBoehme, 2024-02-28:
Thank you for the change, @puterboy.
Just need a new release or RC to test it.

jneilliii, 2024-02-28:
[1.0.4rc3](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/releases/tag/1.0.4rc3) was released with these changes merged

jneilliii, 2024-02-28:
Thanks, updated release notes.

AndreasBoehme, 2024-02-29:
Tested [1.0.4rc3](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/releases/tag/1.0.4rc3), value is correct now!
Thank you for the fast implementation.

closed 2024-02-29 by AndreasBoehme

jneilliii, 2024-02-29:
thanks @puterboy for the fix on that one.

PR #352 (2024-02-09): Fixes	modified:   octoprint_tplinksmartplug/__init__.py
The following pull-request addresses and fixes several bugs/feature enhancements.

Specifically,
- `timestamp` column fixed to be of type `DATETIME` rather than `TXT` #346 
- Extra colunn `grandtotal` added to calculate corrected total net of restarts int `total` column #344
- Entries to `energy_data.db` are only added if: `total` changed, current `power` > 0 or has changed (this captures return to zero) #345
- KWH cost for completed prints is fixed #349

The code covers the following 3 situations:
- New `energy_data.db` creation (one-tim [... cut]