jneilliii/OctoPrint-TPLinkSmartplug #350: [Feature Request]: Toggle between total KWH and cost rather than showing both on same scaled graph

puterboy, 2024-02-08:
Over time if KWH and cost are not the same scale (say cost is $0.20 KWH) then the magnitude difference between the two traces continues to increase linearly making it difficult to see short term changes in energy use.

For example. if:
- KWH=1, then Cost=0.20 and the scale is approximately 0 to 1, so you can perhaps perceive perhaps changes of .05KWH
- KWH=100, then Cost=20 and the scale is approximately 20 to 100, so you can perhaps perceive changes of 5KWH
- KWH=1000, then Cost=200 and the scale is approximately 200 to 1000, so you can perhaps perceive changes of 59KWH
etc.

A way to solve this would be to only display a single trace and add a toggle between them -- this is OK since they basically say the same thing -- i.e. they are parallel lines differing only by the scale parameter $/KWH

puterboy, 2024-02-08:
Even better/easier since assuming constant cost (which I think is all you allow), the curves differ only in scale.
So, why not just have a single curve with different Right and Left hand axis scales -- say left in KWH and right in $.

jneilliii, 2025-09-01:
after looking at the graphing realized that the cost wasn't being assigned to the correct subplot axis, so was getting put on the same graph as Total. this is fixed in the above commit and will be available in the next release candidate.

closed 2026-10-18 by jneilliii