jneilliii/OctoPrint-TPLinkSmartplug #359: [BUG]: Prusa XL fails to start with Puppy Error 17516 when connect on power up is enabled

Anathema-Device, 2024-03-14:
**Describe the bug**
When Octoprint with the TPLINKSmartplug plugin is plugged into a Prusa XL and the TPLinkSmartplug plugin is configured to connect on power-up the printer will fail to boot with "Puppy Error 17516".  Setting the connect time to 60 seconds or more does not resolve the problem.  It can be recovered by going to the printer and pressing reset then back in Octoprint clicking connect when the printer has started.

My guess would be that the USB serial connection is being programmed BEFORE the configured delay and that is causing the issue

**To Reproduce**
Steps to reproduce the behavior:
1. Power off the Prusa XL
2. Connect the USB cable for Octoprint to the back of the machine
3. Configure TPLinkSmartplug to connect after power on - the configured connect delay does not seem to matter
4. Click the lightning bolt in Octoprint to power on the printer
5. After a few seconds the Prusa XL will fail to complete startup and show the message "Puppy Error 17516" on the printer display

**Expected behavior**
At step 5 the printer should start then after the configured delay the USB connection will be configured and connect

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Brave
 - OctoPrint Version : 1.9.3
 - Plugin Version:  1.0.3

**Additional context**
None that I can think of

jneilliii, 2024-03-14:
That error number appears to be related to this knowledge base FAQ: https://help.prusa3d.com/article/puppy-error-17516-xl_400144?

jneilliii, 2024-03-14:
or this forum post: https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-hardware-firmware-and-software-help/puppy-error-17516-when-powering-up-with-octoprint-plugged-in/

Anathema-Device, 2024-03-14:
Yes, I already found the Prusa docs long ago and opened the forum post.  However the problem ONLY occurs if I try to have the auto-connect configured when turning on with the TPLinkSmartPlug plugin regardless of the delay.  When the USB port is enabled it apparently sends a reset to the  port and that results in the printer boot failing.  Seems to me that it could be resolved by not programming the USB connection at all until it is time to connect.

jneilliii, 2024-03-14:
Weird. The plugin doesn't really manage the USB ports themselves, just the process of powering on the plug and doing the connection from OctoPrint. So what you're saying is if the plugin is disabled, the printer turned off, and OctoPrint powered on and connected; when you power on the printer and then go to OctoPrint and hit the connect button it doesn't cause the same error?

Anathema-Device, 2024-03-14:
It is odder than that.  If I configure the plugin to not auto-connect when powered on then I can turn the printer on with the plugin then when it has started click connect in Octoprint and it connects perfectly with the USB cable plugged in all the time.

jneilliii, 2024-03-14:
can you do me a favor and enable debug logging in the plugin settings and restart octoprint. try the process and then when it fails again, share with me the plugin_tplinksmartplug_debug.log, octoprint.log, and screenshots of your settings for the individual plug and the overall setting for the plugin.

closed 2024-04-05 by github-actions[bot]