jneilliii/OctoPrint-TPLinkSmartplug #360: [Feature Request]: Include TP Link Tapo smartplugs

nocluewhatido, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
I unfortunately cant use my TPLink P110 smartplug with octoprint.

**Describe the solution you'd like**
A working TPLink smart plug with octoprint

**Additional context**
I found this unoffical TPLink Tapo API: https://github.com/mihai-dinculescu/tapo would I be possible to add those smartplugs to the octoprint plugin?

jneilliii, 2024-03-15:
I do not plan on integration tapo devices, but someone has already forked this plugin for them.

https://github.com/Tyfy/OctoPrint-TapoSmartplug

closed 2024-03-15 by jneilliii

Hatsune-Cthulhu, 2025-07-23:
Any chance you'll change your stance on this?
TP-Link appear to be sun-setting Kasa in general, so they only sell Tapo devices now